prusa3d/PrusaSlicer #3828: Unable to enumerate HID devices

superm1, 2020-03-12:
### Version
2.2.0-rc2

### Operating system type + version
Chrome OS 81 - Linux.
PrusaSlicer is running in Crostini, which is Debian 10.

### 3D printer brand / version + firmware version (if known)
Creality Ender 3 Pro, connected via octoprint.

### Behavior
* I notice that running PrusaSlicer it is spamming stdout with this error every few seconds:
```
[0x000078cc94020700] [error]   Unable to enumerate HID devices
```
* I just need to open up PrusaSlicer's appimage file and I see the error immediately and continually.
* Since I'm configured to use OctoPrint, i'm not sure why it's even trying to look for local devices.  That being said, Crostini will not show any HID devices available even if I had some.

[image]

_Is this a new feature request?_

No
#### Project File (.3MF) where problem occurs

n/a

bubnikv, 2020-03-12:
Please read the following issue and test with https://github.com/prusa3d/PrusaSlicer/issues/3388#issuecomment-598246157

The message is about enumerating human interface devices, namely the 3DConnexion mouse is searched for.

superm1, 2020-03-12:
Thanks for responding.  I downloaded and ran that build (with bumping up loglevel to 4 as one of the most recent posters did).  Here's the output:

```
./PrusaSlicer-2.2.0-rc3+15-linux-x64-g6866c2b-202003121402.AppImage\?dl\=0  --loglevel=4
[2020-03-12 10:35:28.424201] [0x000078566676a540] [info]    Checking if indices need to be installed from resources...
[2020-03-12 10:35:29.493648] [0x000078566676a540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 69MB; Shared memory: 52MB; Private memory: 17MB; Peak memory usage: 69MB
[2020-03-12 10:35:31.505922] [0x00007856610d7700] [error]   Unable to enumerate HID devices
[2020-03-12 10:35:32.283938] [0x000078566676a540] [debug]   TriangleMesh::repair() started
[2020-03-12 10:35:32.824717] [0x000078566676a540] [debug]   TriangleMesh::repair() finished
[2020-03-12 10:35:33.165712] [0x000078566676a540] [debug]   TriangleMesh::repair() started
[2020-03-12 10:35:33.166381] [0x000078566676a540] [debug]   TriangleMesh::repair() finished
[2020-03-12 10:35:33.507810] [0x00007856610d7700] [error]   Unable to enumerate HID devices
[2020-03-12 10:35:34.171638] [0x000078566676a540] [info]    Checking for cached configuration updates...
[2020-03-12 10:35:34.171957] [0x000078566676a540] [info]    Confing bundle not installed for vendor PrusaResearch, skipping: 
[2020-03-12 10:35:34.172239] [0x000078566676a540] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2020-03-12 10:35:34.172402] [0x000078566676a540] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2020-03-12 10:35:34.173640] [0x000078566676a540] [debug]   Vendor: Creality, version installed: 0.0.2, version cached: 0.0.2
[2020-03-12 10:35:34.173662] [0x000078566676a540] [info]    No configuration updates available.
[2020-03-12 10:35:34.212667] [0x00007856592e5700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
[2020-03-12 10:35:35.143332] [0x00007856592e5700] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-03-12 10:35:35.145737] [0x00007856592e5700] [info]    Syncing configuration cache
[2020-03-12 10:35:35.145803] [0x00007856592e5700] [warning] No such vendor: PrusaResearch
[2020-03-12 10:35:35.145819] [0x00007856592e5700] [warning] No such vendor: BIBO
[2020-03-12 10:35:35.145830] [0x00007856592e5700] [warning] No such vendor: LulzBot
[2020-03-12 10:35:35.145841] [0x00007856592e5700] [info]    Downloading index for vendor: Creality
[2020-03-12 10:35:35.145899] [0x00007856592e5700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/Creality//index.idx`
        -> `/home/superm1/.PrusaSlicer/cache/Creality.idx-update`
        via tmp path `/home/superm1/.PrusaSlicer/cache/Creality.idx-update.4072.download`
[2020-03-12 10:35:35.510905] [0x00007856610d7700] [error]   Unable to enumerate HID devices
[2020-03-12 10:35:36.171475] [0x00007856592e5700] [debug]   Got index for vendor: Creality: current version: 0.0.2, recommended version: 0.0.2
[2020-03-12 10:35:37.514747] [0x00007856610d7700] [error]   Unable to enumerate HID devices
[2020-03-12 10:35:39.518570] [0x00007856610d7700] [error]   Unable to enumerate HID devices

```

bubnikv, 2020-03-12:
It looks like the error message is spurious in your case as you likely do
not have any USB HID (Human Interafec Device, for example joystick or
3DConnexion mouse) connected. We should fix that.

bubnikv, 2020-03-12:
By the way, do you have any USB keyboard, mouse, joystick or any other HID (Human Interface Device) connected to your computer? If so, then you should not get this error message. In that case you exhibit a similar issue to #3388 indeed.

superm1, 2020-03-12:
On ChromeOS no USB devices are passed into Crostini, so even though I do have a USB mouse plugged in the kernel in Crostini won't see it.  I checked on a second machine (XPS 7390) running Ubuntu 20.04 daily image.  This one does have a USB mouse and logitech unifying receiver plugged.

I do see the same messages about being unable to enumerate HID devices there too.

bubnikv, 2020-03-15:
This spurious error message will be removed in PrusaSlicer 2.2.0-rc4. Closing.

closed 2020-03-15 by bubnikv

superm1, 2020-03-16:
Thanks, confirmed the fix!