prusa3d/PrusaSlicer #3830: 2.2.0-rc3 - right click to repair with Netfabb does nothing while the model is selected

keksa, 2020-03-12:
### Version
2.2.0-rc3

### Operating system type + version
Windows 10 Pro, Version 1903 (OS Build 18362.720)

### 3D printer brand / version + firmware version (if known)
probably irrelevant, but Prusa Mini 4.0.4

### Behavior
1) add a broken model on the build plate
2) little warning icon appears in the object list in right sidebar -> tooltip says "right click to repair with Netfabb"
3) select the model by left clicking on it on build plate, or by left clicking on it in the sidebar list, does not matter
4) right click the warning icon & nothing happens (not a misclick, tried several times, same result)
5) deselect the model by clicking elsewhere on the build plate
6) right click the icon & everything works correctly (repair starts)

_Is this a new feature request?_
no

#### Project File (.3MF) where problem occurs
I can upload one if you really need to, but it's happening for literally every broken object, just pick something on thingiverse.

bubnikv, 2020-03-13:
Is it a new issue? Did it work in PrusaSlicer 2.2.1?
I quickly tested, it works for me.

Did you deinstall any of the Microsoft 3D printing applications? Maybe that
deinstalled the Netfabb stuff?

keksa, 2020-03-13:
@bubnikv Hi, yes it's a new issue. Just tried and works correctly in latest stable (2.1.1) and also in 2.2.0-rc and 2.2.0-rc2. It's only the last one (2.2.0-rc3) where this is a problem.

Also I think you misunderstood, it still works in 2.2.0-rc3, but only **after** deselecting the model.

keksa, 2020-03-13:
@bubnikv I did a video - https://youtu.be/fPL73j_xVsE when it seems I'm just hovering on the icon, that's me actually right clicking.

bubnikv, 2020-03-13:
Verified that the left mouse button no more works in PrusaSlicer 2.2.0-rc3. We are investigating further where it broke.

bubnikv, 2020-03-13:
The following commit broke it 4ca03c3f8a7d0013c65197c5b93b81c95c9d72f6
when trying to fix https://github.com/prusa3d/PrusaSlicer/issues/3802

bubnikv, 2020-03-14:
This is a regression, broken in PrusaSlicer 2.2.0-rc3. It will be fixed in rc4. Also selection of extruder was broken on OSX.

Thanks for heads up. Closing.

closed 2020-03-14 by bubnikv