prusa3d/PrusaSlicer #3831: Support interface isn't generated: build plate only + blocked by model + support enforcer

rcmaniac25, 2020-03-13:
### Version
2.1.1+win64
PrusaSlicer-2.1.1+win64-201912101512

### Operating system type + version
Operating System: Windows
System Architecture: 64 bit
Windows Version: Windows 10 (build 18363), 64-bit edition

### 3D printer brand / version + firmware version (if known)
Prusa Mini, stock, 4.0.4 (though no printing has been done, as the issue is during slicing)

### Behavior
Went to print a model (https://www.thingiverse.com/thing:4166787) and there are two overhangs that are above one another. As I didn't need supports everywhere, I specified "Support on build plate only"
[image]
You can see, it's printing in mid-air. This is expected for this supports method + model blocking the 2nd overhang's support:
[image]

I added a support enforcer without changing the supports settings:
[image]
...which resulted in supports for the overhang:
[image]
...but, there's no support interface (neighboring overhangs shown for comparison):
[image]
...they show up later, but in the same spots as prior to the support enforcer:
[image]

With support enforcers only:
[image]

Supports: Everywhere (no support enforcer):
[image]

I expect that with a support enforcer, even "Support on build plate only" will generate proper supports and not miss the support interface or place the interface only where the build plate is exposed.

#### Project File (.3MF) where problem occurs
[issue3831.zip](https://github.com/prusa3d/PrusaSlicer/files/4327966/issue3831.zip)

Seems related to #2801, but that issue seems to be "supports not generated" as opposed to "supports is generated, but the interface is not"

kursatu, 2020-05-05:
This has been around for few years , as far as I can remember.
Can this be finally fixed?
It greatly diminishes the value of multi-material printing.
Here is my model which has the missing bottom interfaces marked with red.
The top interface layer is generated and printed fine.
 
[image]
[image]

togomedia, 2020-12-06:
Major issues with Support Enforcer. Does not function correctly in many circumstances.

bubnikv, 2021-02-19:
The supports are blocked by "from build plate only" even for support enforcers.

rcmaniac25, 2021-02-23:
Sure... except when adding the support enforcer, supports do get generated. But the support interface does not. So I'd expect it to not generate any supports based on what you've stated, but it does generate them. (note: going off my images, I'd need to test this in 2.3.0 to see if behavior changed)

bubnikv, 2021-03-18:
Duplicate of #1997 #2801

Reworked to give the support enforcers precedence over "on build plate only".

[image]

[image]

rcmaniac25, 2021-03-19:
Fantastic

bubnikv, 2021-03-19:
Fixed with 9f09f03228478aa77549efec845717b06e46cbc3

closed 2021-03-19 by bubnikv