prusa3d/PrusaSlicer #3832: Slicer generates empty layers from OpenSCAD model

mediaDS, 2020-03-13:
### Version
2.2.0-rc3

### Operating system type + version
Linux Mint 19.3 64-Bit

### 3D printer brand / version + firmware version (if known)

Anycubic i3 Mega

### Behavior

After modelling an object in OpenScad, I always get an "empty layer" error. It is an object of two cubes, one rotated 90°. As soon as I put holes into one of the cubes (before rotating), I will get an error message. If I use only the plain cubes without holes, then everything is ok.

The 3D-view of Slicer dispalyes the correct object, but the print preview shows a broken object. See screenshots.

I imported the resulting .stl file with Freecad and Cura without issues. Only Prusa Slicer gies the error message.

**Screenshots**

After Import:

[image]

3D-View seems ok:

[image]

3D Print Preview does not look ok:

[image]

The object does work in Freecad (and Cura) without issues:

[image]

Long story short - I do not have any idea what could cause the problem. It may be either OpenScad or the Slicer. Please ask if you have any further questions.

#### Project File (.3MF) where problem occurs

[Slicer_Failed.zip](https://github.com/prusa3d/PrusaSlicer/files/4329362/Slicer_Failed.zip)

espr14, 2020-04-14:
The STL is not valid 3D object. The missing part has 0 volume. In PrusaSlicer right-click on the object and select "Fix through the Netfabb". This will fix the issue. OpenScad has problem with joining two object that are touching each other. You have to move one to have nonempty intersection. If you send me the .scad file I can show you.

The black top area are flipped faces:
[image]

After Netfabb fix:
[image]

FidelCapo, 2021-02-01:
This issue should be fixed in next release.
Closing.

closed 2021-02-01 by FidelCapo