prusa3d/PrusaSlicer #3833: PrusaSlicer fails to slice when z height is larger than ~2150 mm

vmario89, 2020-03-13:
### Version
2.1.1+win64-201912101512

### Operating system type + version
Win10 x64 Pro

### 3D printer brand / version + firmware version (if known)
DIY

### Behavior
When entering a value larger than approx 2150mm for z height, the slicer won't slice
[image]. The "slice now"  button just does nothing. Reducing the value to 2140 works for me.

Why did i find this out? Why do i need this?
I am building a really large machine. Now i am trying to configure the slicer profile for it to test the printer over complete height. my z-max is about 2500 mm.

FidelCapo, 2020-03-13:
I think this crash for the same reason as https://github.com/prusa3d/PrusaSlicer/issues/3502

bubnikv, 2020-03-13:
the crash is due to 32bit resolution of fixed point coordinates. This was a
decision I made 4 years ago when 32bit operating systems were still
relevant. It is time to re-evaluate PrusaSlicer with 64bit fixed point
coordinates.

bubnikv, 2020-03-15:
duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3502

closed 2020-03-15 by bubnikv