prusa3d/PrusaSlicer #3834: "No sparse layers" on purge block/wipe leaves a midair artifact

VanessaE, 2020-03-13:
### Version
git commit 45e00793567782b0103bca0bc2e266be708a32a2

### Operating system type + version

Debian sid

### 3D printer brand / version + firmware version (if known)

Hypercube with two extruders feeding a Y splitter (NOT MMU).

### Behavior

Setting "no sparse layers" on the purge block creates a single, midair purge layer above the block at approximately the "normal" height of the block.

Normal mode:

[image]

"No sparse layers" mode:

[image]

(yeah, I know the block is large compared to the model.  I've only just started experimenting with multi-color printing and have not tuned the filament switches well)

#### Project File (.3MF) where problem occurs

[3DBenchy_-_Dualprint.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4330563/3DBenchy_-_Dualprint.3mf.zip)

lukasmatena, 2020-03-14:
This is intentional. See whichever of [FAQ](https://github.com/prusa3d/PrusaSlicer/wiki/Frequently-Asked-Questions-(FAQ)), #3593, #3533, #3116, #2412, #1355, #2088, #1616, #1070, #1268, #1423, #1386, #663, #1581 or #2612.

Please, search the tracker before filing a new one issue. Closing.

VanessaE, 2020-03-16:
@lukasmatena OK, so one floating top layer might it intentional, but surely THIS is not:

[image]

(this is 2.2.0-rc4 with the same dual-print benchy mentioned earlier)

bubnikv, 2020-03-16:
I am not seeing it with your 3MF

[image]

Just a wild guess, but it may be a floating point numeric issue,

bubnikv, 2020-03-16:
it may be an OpenGL driver issue (not enough GPU memory?)
or some slicing service update issue after parameter changes?

The last one is easily ruled off if you yourself cannot reproduce the issue after saving a 3MF, closing slicer, reopening slicer with that 3MF and reslicing. If the issue is not there, than the issue is likely caused by the slicing core update by a parameter change.

bubnikv, 2020-03-16:
I don't have your fancy print bed texture. Does the issue disappear if you disable the print bed texture? That would indicate low GPU memory issue.

VanessaE, 2020-03-16:
My GPU has 3GB, and can handle far more graphics detail than is shown there.  Besides, it works in 2.2.0-rc3, and in fact, I had several 2-color parts on the plate under 2.2.0-rc3 to test something.  There was around 5 times as much on display, and it was fine.

I also tried just loading the original Benchy files instead of the 3mf.  Same issue.

In fact, on watching closely: I can see it render the entire purge block as it would exist if "no sparse layers" were turned off.  A moment after the full block appears, it then erases the parts that wouldn't be needed, it just doesn't "compact" it down.

I then exported the broken-looking result to g-code, then loaded it into Pronterface:

[image]

Since it looks the same in Pronterface, it is clearly a toolpaths/g-code issue, NOT a GPU memory issue.

bubnikv, 2020-03-16:
rc4 self compiled or our AppImage?
Do you see it with our AppImage as well?

VanessaE, 2020-03-16:
I'm using the official rc4 appimage.

VanessaE, 2020-03-16:
If it matters, here is my configuration (from File -> Export -> Export Config):

[Prusaslicer-config-20200316.ini.zip](https://github.com/prusa3d/PrusaSlicer/files/4337917/Prusaslicer-config-20200316.ini.zip)

Something I noticed: I could swear that I had specific purge volumes set when I saved that 3mf, but if I did, they were not applied on-load.

VanessaE, 2020-03-16:
Well I'd say that rules-out anything specific to my setup :grin:

bubnikv, 2020-03-16:
I don't see it running on Linux in virtual box either.

[image]

VanessaE, 2020-03-16:
But I do note an issue on yours, @bubnikv:  the gunwale and other details are supposed to be *orange* in that 3mf.  Not white.

VanessaE, 2020-03-16:
I tried removing the texture and model from the bed and just setting it back to a plain rectangle, 200x200 size.  No change.

*keeps tinkering*

AH HAH!

I'd forgotten about a workaround I had to add to the tool change g-code:
```
; "tool change" code goes here

; force the correct temp for the hotend after toolchange
{if current_extruder == 0}
    M104 T0 S[temperature_0]
    M104 T1 S[temperature_1]
{elsif current_extruder == 1 }
    M104 T1 S[temperature_1]
    M104 T0 S[temperature_0]
{endif}

; force Z to stay at the proper height after toolchange
T[next_extruder]
G1 Z[layer_z]

; end of "tool change" code
```

Deleting those last two commands is enough to make the purge tower work, BUT they're a workaround for #3848.

bubnikv, 2020-03-16:
now that explains everything. Would you please attach the 3mf with your
workaround integrated, so that we can retest?

VanessaE, 2020-03-16:
This should do:  [3DBenchy_-_Dualprint, 2.2.0-rc4 with Z workaround.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4338918/3DBenchy_-_Dualprint.2.2.0-rc4.with.Z.workaround.3mf.zip)

Just to be clear, I started RC4, added the 2-part Benchy from the original files, set the two filaments, sliced, saved the g-code to a temp file (not that this part matters), then saved the 3mf project file.

lukasmatena, 2020-03-16:
Ok, so what is the problem? Your custom g-code tells the printer to move to some height.... and it moves to that height. What are we supposed to do?

VanessaE, 2020-03-17:
Did you read #3848?  Without the above workaround, which obviously doesn't work well enough :confused:, the first E1 usage is at the wrong height, which luckily so far has resulted merely in poor bed adhesion of that tool's filament due to it just loosely being laid down from too high.

lukasmatena, 2020-03-17:
I did read #3848 and I don't know yet how justified the claim is and whether any workaround is or isn't needed. But that is another issue.

Now that I think about it a bit, you are probably right that the `layer_z` placeholder should contain actual z-height of the toolchange. I think that should indeed be fixed.

lukasmatena, 2021-03-05:
@VanessaE There is a new placeholder `toolchange_z` available in custom toolchange gcode since c25c435. It should allow you to do what you need. The `layer_z` still contains the z of the print that triggered the toolchange, the new one contains the actual z where the toolchange takes place.

Does it solve your issue?

bubnikv, 2021-06-03:
implemented on master with c25c435. Closing.

closed 2021-06-03 by bubnikv