prusa3d/PrusaSlicer #3835: Add/Remove Filaments is not hiding all of the un-checked filaments

jhenley01, 2020-03-13:
2.2.0 RC3 

MacOS Mojave 10.14.5

Prusa Mini 4.0.4

The option for Add/Remove Filaments in the filament dropdown is not hiding all of the un-checked filaments.

See screenshots of the filaments I have selected vs what is displayed in the dropdown.

[image]
[image]

bubnikv, 2020-03-14:
Please zip and attach the content of your PrusaSlicer configuration
directory if you can.

jhenley01, 2020-03-14:
> […](#)

[PrusaSlicer.zip](https://github.com/prusa3d/PrusaSlicer/files/4332386/PrusaSlicer.zip)

Hopefully this is what you need, I pulled it from ~Library⁩/Application Support⁩/PrusaSlicer

bubnikv, 2020-03-14:
Thanks. I see the issue. Update from 2.1.1 to 2.2.0 is no clean in regard to the list of installed filaments.

jhenley01, 2020-03-14:
> Thanks. I see the issue. Update from 2.1.1 to 2.2.0 is no clean in regard to the list of installed filaments.

Yes that was it, did a clean install and that fixed the issue. Thank you!

bubnikv, 2020-03-15:
Conversion of profiles from PrusaSlicer 2.2.1 to 2.2.0 will be fixed in PrusaSlicer 2.2.0-rc4.
Closing.

closed 2020-03-15 by bubnikv