prusa3d/PrusaSlicer #3837: Wipe into object causes hang/crash

VanessaE, 2020-03-14:
### Version

git commit 45e00793567782b0103bca0bc2e266be708a32a2

### Operating system type + version

Debian sid

### 3D printer brand / version + firmware version (if known)

Hypercube with two extruders feeding a Y splitter to one hotend

### Behavior

Trying to enable the wiping into an object causes one of two errors, depending on which mode you're using.

**First error:**

1. Enable a wipe tower
2 Add three objects to the plate
3. Make two of them print with different extruders.
4. On the third object, click the "gear" in the right sidebar to get the object's settings menu.  
5. Select "Add settings" -> "Wipe options".
6. Tick "Wipe into this object's infill" and click "OK"

A window briefly opens, immediately followed by:

[image]

You can also get this error to pop up if you just select "Wipe options" from the top level of the menu (i.e. without first clicking "Add settings".  A backtrace is provided.

<details><summary><b>Show backtrace...</b></summary>

```
ASSERT INFO:
../src/gtk/bitmap.cpp(917): assert "IsOk()" failed in GetHeight(): invalid bitmap

BACKTRACE:
[1] wxBitmap::GetHeight() const
[2] wxDataViewCustomRendererBase::WXCallRender(wxRect, wxDC*, int)
[3] gtk_cell_renderer_render
[4] gtk_container_propagate_draw
[5] gtk_container_propagate_draw
[6] g_closure_invoke
[7] g_signal_emit_valist
[8] g_signal_emit
[9] gtk_container_propagate_draw
[10] gtk_container_propagate_draw
[11] g_closure_invoke
[12] g_signal_emit_valist
[13] g_signal_emit
[14] gtk_container_propagate_draw
[15] g_closure_invoke
[16] g_signal_emit_valist
[17] g_signal_emit
[18] gtk_container_propagate_draw
[19] gtk_container_propagate_draw
[20] g_closure_invoke
[21] g_signal_emit_valist
[22] g_signal_emit
[23] gtk_container_propagate_draw
[24] gtk_container_propagate_draw
[25] gtk_main_do_event
[26] g_signal_emit_valist
[27] g_signal_emit
[28] g_main_context_dispatch
[29] g_main_loop_run
[30] gtk_main
[31] wxGUIEventLoop::DoRun()
[32] wxEventLoopBase::Run()
[33] wxAppConsoleBase::MainLoop()
[34] wxEntry(int&, wchar_t**)
[35] __libc_start_main

```
</details>

If I click "Stop", Slic3r closes, of course.  If I click "Continue", I am presented with an empty window and Slic3r hangs, spinning at 100% of one CPU core.  Sometimes the empty window is literally just a frame, showing my desktop's contents under it, live.  Sometimes it's one of the more common kinds of hangs where it'll show distorted, static copies of other parts of my desktop.

Sometimes, that empty window will immediately disappear, to be replaced by a repeat of the above error dialog.

Sometimes, I just get the empty window, without first getting the error dialog.

Clicking "Wipe options" in the top-level menu (I.e. without going through the "Add settings" submenu) will also produce the above error and/or the empty window.

Trying to close the empty window does nothing for a bit, then trigger's my desktop's "this window is not responding" watchdog.  Terminating the empty window kills Slic3r.

Included below is a project file that'll get you most of the way to this error.  Just load it, and add something small like a cube or a calicat.

#### Project File (.3MF) where problem occurs

[3DBenchy_-_Dualprint_E1_first.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4332224/3DBenchy_-_Dualprint_E1_first.3mf.zip)

lukasmatena, 2020-03-14:
What wxWidgets/GTK versions have you linked against? Do you have the same issue with the official 2.2.0-rc3 AppImage?

VanessaE, 2020-03-14:
wx 3.0 as supplied in Debian sid.  I don't use appimages.

VanessaE, 2020-03-15:
But just to see what happens, I tried the 2.2.0-rc3 appimage.  No problems there, the wipe-into feature works fine.

VanessaE, 2020-03-19:
Incidentally I also get this error when built against wx 3.1 as supplied from codelight.org's Debian repository.

VanessaE, 2020-03-20:
Ok, this actually affects other items in the settings menu, not just the wipe options.
 
HOWEVER:  I was able to hack my way around it.  I run it under gdb and was able to follow the code back to a point where I could at least insert a workaround.

I altered `src/slic3r/GUI/ObjectDataViewModel.cpp`, function  `BitmapChoiceRenderer::Render()` at line 1702, to read as follows:

```c++
bool BitmapChoiceRenderer::Render(wxRect rect, wxDC* dc, int state)
{
    int xoffset = 0;
    int htmp = 0;

    const wxBitmap& icon = m_value.GetBitmap();

    if (icon.IsOk() && icon.GetHeight())
    {
        htmp = icon.GetHeight();
    }

    if (icon.IsOk())
    {
        dc->DrawBitmap(icon, rect.x, rect.y + (rect.height - htmp) / 2);
        xoffset = icon.GetWidth() + 4;
    }

    if (rect.height==0)
        rect.height= htmp;
    RenderText(m_value.GetText(), xoffset, rect, dc, state);

    return true;
}
```
(that is, created a variable `htmp` and used it to pass on the value from icon.GetHeight(), if any)

I'm pretty sure it's technically "wrong", but I can now turn the option on and see its effect in the preview, so I guess the actual feature is working. Comparing this hacked version to RC4, I see no appreciable difference in the GUI -- they look the same to me (my tweak is on the left, obviously):

[image]

bubnikv, 2020-03-20:
@YuSanka would you please review this proposal?

VanessaE, 2020-03-20:
I wouldn't call it a "proposal" so much as "quick and nasty hack" :smile:  I'm sure it can be done more cleanly (like roll the first if() into the second one).

dakkar, 2020-04-06:
I've been having the same problem. I'm writing ebuilds (Gentoo Linux packages) for prusa-slicer, and I've gone through these combinations:

* prusa-slicer 2.1.1 built against Gentoo's wxWindows 3.0.4 (GTK 2, some patches) works
* prusa-slicer 2.2.0 built against the same xwWindows crashes as described here
* prusa-slicer 2.2.0 build against wxWindows 3.1.3 (my own package, no patches) crashes as described here
* pre-built prusa-slicer 2.2.0 (which is probably statically linked against wxWindows 3.1.1 plus some MacOS patches from prusa's fork) works

Another difference: my slicer is built with `-DSLIC3R_FHS=1`

I can compile nearly any combination… how can I help?

dakkar, 2020-04-06:
I *think* the problem started with 
https://github.com/prusa3d/PrusaSlicer/commit/d7ebc4de5baca66df1820627222a804caa2568df#diff-b3f8c44e1c9fadbc60ddf42bf5ceb631R2124
notice that the `rect.height= icon.GetHeight();` is outside the `if (icon.IsOk())`

dakkar, 2020-04-06:
https://www.thenautilus.net/cgit/gentoo-overlay/tree/media-gfx/PrusaSlicer/files/PrusaSlicer-2.2.0-fix-get-height-crash.patch this patch fixes this particular crash for me.

lukasmatena, 2020-05-07:
@dakkar 's PR was merged, hopefully fixing this problem. Thanks for the fix.

PR #4006 (2020-05-07): avoid calling GetHeight() on non-IsOk() icons
probably fixes #3837
commit 34b504d5 (2020-05-07): avoid calling GetHeight() on non-IsOk() icons

probably fixes #3837
closed 2020-05-07 by lukasmatena