prusa3d/PrusaSlicer #3839: [FR] An object larger than the plate could still be printable

VanessaE, 2020-03-14:
### Version

git commit 45e00793567782b0103bca0bc2e266be708a32a2

### Operating system type + version

Debian sid

### 3D printer brand / version + firmware version (if known)

Any make and/or model where the theoretical volume is horizontally larger than the bed (Prusa i3 and most other "bedflingers", for example).

### Behavior

Load anything with a smaller bottom surface than the X/Y bounds of the whole object, such as a Benchy or a bowl.  Scale it up until Slic3r thinks it won't fit:

[image]

The program will refuse to slice the object.

While I have no reason to print this overgrown tugboat, it should work just fine in my Prusa i3.

Like most bedflingers, a Prusa i3 has more considerably Y travel than is strictly necessary for the nozzle to reach the whole print surface, so it's possible to print something bigger than the bed, as long as the first layer or two still fits.

Although this mostly applies to bedflingers, it would also work on my Hypercube, though to a much smaller degree (that machine has only a couple of millimeters of extra Y travel, and about 12 mm of extra along X).

I suggest the following:

* Let the user slice the object anyway, regardless of its apparent fit, as upstream Slic3r would.
* At the bottom of the window, change the error message into a warning, so that it reads something like:
```
  An object outside the build plate was detected
  The sliced result may not be printable.
```
 * When the user slices the plate's contents, detect if any part of the first layer (or perhaps the first two or three layers) is off-plate, THEN throw an error if appropriate.  Otherwise, just keep the above warning message at the bottom of the sliced preview.

It may be prudent to let the user tell Slic3r about the theoreticall absolute max volume.

I suppose one could specify a larger bed size than reality, and then load an appropriately-prepared bed model and/or overlay image that make it clear where the actual print surface is in relation to the over-travel areas, but that's a kludge, and wouldn't allow Slic3r to warn the user about objects that really wouldn't fit (first layer or not).

#### Project File (.3MF) where problem occurs

N/A, just load a Benchy and scale it up.

bubnikv, 2020-03-14:
You may increase the build volume by defining the print bed as large a your
print volume, and then you may paint your print bed area into the print bed
texture. This way PrusaSlicer will not warn you about the print bed
violation the same way the upstream slic3r will not.

VanessaE, 2020-03-14:
Right, but as I said, that's a workaround that would bypass some sanity checks i.e. the program would be able to warn me if the part is totally out of the build volume, but not if its bottom is off-plate.

closed 2026-09-29 by prusaslicer-bot[bot]