prusa3d/PrusaSlicer #3840: Elephant Foot Compenstation affecting thin walled objects

jweaver100, 2020-03-14:
### Version
2.2.0-RC3

### Operating system type + version
Windows 7 & 10

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
After upgrading to 2.2.0 I tried to slice/print an object that I had printed countless times and the first layer failed.

Here is the object in question:
**https://www.thingiverse.com/thing:3653631**

After watching it closely, the layer was being laid down in an unexpected order leaving open ends which were picked up when the extruder passed. 

I studied the tool path in a gcode viewer and compared to some sliced files that I had before and the path is very different in 2.2.0

Where as in previous releases, the inner, then outer perimeter would go down and then the fill all in a very logical/linear order.. But in 2.2.0 it was all over the place, laying part of the inner.. Part of the outer.. Then the fill and so on..

After discussing this on the Prusa forum I was given a tip off that Elephants Foot Compensation might be the cause, as its set to 0.2mm in Slic3r.. And true enough.. When I set it to 0mm, Slicing returned to normal.

So it seems that Elephants Foot compensation has problems with thin walled objects, and since Slic3rs default is now 0.2mm people might struggle getting certain objects to print.

#### Project File (.3MF) where problem occurs

[elephantsfootdemo.zip](https://github.com/prusa3d/PrusaSlicer/files/4332462/elephantsfootdemo.zip)

bubnikv, 2020-03-14:
1st layer

[image]

2nd layer

[image]

The geometry seems all right to me. The unexpected order of extrusions is due to the gap fill kicking in and this did not change compared to PrusaSlicer 2.2.1.

We are working on improving the perimeter generator and gap fill for the next major release, not for PrusaSlicer 2.2.

jweaver100, 2020-03-14:
Something has changed for sure.. I am Slicing this 3 ways.

(BTW I am not talking about the layer view in Slicer I am talking about the order that its prints, seen either on the printer, or via a GCODE viewer).

1. With 2.1 - Perfect.. Complete Inner perimeter goes down, then outer.. and then fill
2. With 2.2.0 and default settings (0.2mm elephants foot) - 1st layer not right.
3. With 2.2.0 and 0mm elephants foot - Perfect.. Same as 2.1

As soon as I enable Elephants Foot compensation, the order that the layer goes down is very odd...

jweaver100, 2020-03-14:
Here are some pictures from the GCODE viewer... This is from 2.1 or with EF set to 0mm

[image]

And now when EFC = 0.2mm

[image]

I don't know if this shows what I am trying to explain...

jweaver100, 2020-03-16:
I had a play with this and can see exactly whats happening.. Because EFC reduces the size of the bottom layer, this means that thin walled objects get thinner and sometimes too thin to print. In my case, it makes the small details on the bottom layer shrink, altering the order that they are printed.

So for any object where you have small details on the first layer this is like to have an unpleasant effect, as EFC will erase some of the detail...

I don't know what the solution is for this.. Other than detecting small details/think walls and disabling EFC for these objects... Perhaps there needs to be a warning, or perhaps even a button (like Brim) where you can turn EFC on/off, rather than dig down in the settings for it.

Personally I think I am just unlucky hitting this problem with this particular object.. But I do think that this is something others are going to fall foul off when printing objects with fine details.

bubnikv, 2020-03-16:
> Other than detecting small details/think walls and disabling EFC for these objects.

That is what we do in PrusaSlicer 2.2.0. The object is not shrunk below two perimeter lines. You may test it by shrinking your object. You will see that the elephant foot compensation effect will be weaker as the 1st layer thin wall gets thinner.

jweaver100, 2020-03-16:
That makes sense... the problem with this object in particular as that the first layer shrinks enough to mean the clips/bulges are layed down out of order.. And stray ends are left until later where they are ripped up

The bottom line is that with 2.1 and 2.2.0 with EFC turned off, this object lays down exactly as you might expect.. But as soon as EFC is enabled, it doesn't lay down in a logical order and the 1st layer is horrible.

bitingsock, 2022-08-16:
> That is what we do in PrusaSlicer 2.2.0. The object is not shrunk below two perimeter lines.
@bubnikv Do you know why this was chosen? why not one line? That said, do you know anyway to force this limit?

closed 2026-09-29 by prusaslicer-bot[bot]