prusa3d/PrusaSlicer #3842: Wipe to Infill causes severe overextrusion (thicker walls & infill)

jettoblack, 2020-03-15:
### Version
2.2.0-beta or 2.1.0-release

### Operating system type + version
Windows 10

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMU2S

### Behavior
I hesitated to post this bug report as you'd think others would have seen this too, but given that the EXACT SAME 3MF file causes such a difference simply by toggling Wipe to Infill enabled/disabled shows me the problem has something to do with this software setting and not a hardware or filament issue. (I can reproduce this with any filament and several different models.) Perhaps one of my other slicer settings is set to something unusual and causing this result, but if so, I don't know what.

When "Wipe to Infill" is enabled, perimeters and infill lines are severely overextruded, even on parts on the plate which have a single material and no wiping done in their infill. You can see in the pictures below that the single material object has drastically different thickness when using wipe to infill enabled vs. disabled. On some models, so much extra plastic is extruded that it builds up in Z and results in nozzle crashes and print failures. Using the exact same settings other than setting "Wipe to infill" disabled does not show the problem.

[image]
[image]
[image]
[image]
[image]
[image]

#### Project File (.3MF) where problem occurs

[wipe to infill disabled.zip](https://github.com/prusa3d/PrusaSlicer/files/4333690/wipe.to.infill.disabled.zip)
[wipe to infill enabled.zip](https://github.com/prusa3d/PrusaSlicer/files/4333691/wipe.to.infill.enabled.zip)

VanessaE, 2020-03-16:
2.2.0-rc also exhibits this problem. :frowning_face:

VanessaE, 2020-03-16:
2.2.0-rc also exhibits this problem. :frowning_face: 

Isn't there some kind of workaround?  Purge towers are wasteful, especially if you use a "high flow" hotend that has a large melt chamber (and thus needs a stupidly-large purge volume when the colors are too dissimilar), but at least purging into infill can help temper that.

Also, OCD time:

Please consistently name this feature: flushing the nozzle is routinely called purging here and there in and outside of Prusaslicer, but the object onto which all that filament is deposited is confusingly referred to as a WIPE tower.  "Wipe" is a totally different action: brushing the nozzle around in an object's infill without extruding, to try to wipe-off any ooze prior to a travel move.  Almost none of the moves that take place on the purge tower are wiping actions under that definition.  :grin:

jettoblack, 2020-03-16:
I can confirm this problem still occurs in 2.2.0-RC4. Attached is gcode generated by RC4 and it's still printing but I can already see the problem.

[PS220-rc4 wipe to infill enabled_0.2mm_PLA_MK3SMMU2S_1h56m.zip](https://github.com/prusa3d/PrusaSlicer/files/4338669/PS220-rc4.wipe.to.infill.enabled_0.2mm_PLA_MK3SMMU2S_1h56m.zip)

bubnikv, 2020-03-17:
Hopefully fixed, you may test with https://www.dropbox.com/s/bro06md0igzwgz9/PrusaSlicer-2.2.0-rc4%2B11-win64-g98fb8b626-202003171059.zip?dl=0

VanessaE, 2020-03-17:
Ew, a windows build. For shame. :grin:

bubnikv, 2020-03-18:
It will be fixed in PrusaSlicer 2.2.0-rc5. Closing.

closed 2020-03-18 by bubnikv