prusa3d/PrusaSlicer #3845: Build plate rendering not working

ercoleg, 2020-03-15:
Version:   2.2.0-rc4+win64
Build:     PrusaSlicer-2.2.0-rc4+win64-202003151137

Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 8,467MB
OpenGL installation
GL version:   4.5.0 NVIDIA 382.05
Vendor:       NVIDIA Corporation
Renderer:     NVS 4200M/PCIe/SSE2
GLSL version: 4.50 NVIDIA
Installed extensions:

GL_AMD_multi_draw_indirect
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
GL_ARB_gpu_shader_int64
GL_ARB_half_float_pixel
GL_ARB_half_float_vertex
GL_ARB_imaging
GL_ARB_indirect_parameters
GL_ARB_instanced_arrays
GL_ARB_internalformat_query
GL_ARB_internalformat_query2
GL_ARB_invalidate_subdata
GL_ARB_map_buffer_alignment
GL_ARB_map_buffer_range
GL_ARB_multi_bind
GL_ARB_multi_draw_indirect
GL_ARB_multisample
GL_ARB_multitexture
GL_ARB_occlusion_query
GL_ARB_occlusion_query2
GL_ARB_parallel_shader_compile
GL_ARB_pipeline_statistics_query
GL_ARB_pixel_buffer_object
GL_ARB_point_parameters
GL_ARB_point_sprite
GL_ARB_program_interface_query
GL_ARB_provoking_vertex
GL_ARB_query_buffer_object
GL_ARB_robust_buffer_access_behavior
GL_ARB_robustness
GL_ARB_sample_shading
GL_ARB_sampler_objects
GL_ARB_seamless_cube_map
GL_ARB_separate_shader_objects
GL_ARB_shader_atomic_counter_ops
GL_ARB_shader_atomic_counters
GL_ARB_shader_bit_encoding
GL_ARB_shader_draw_parameters
GL_ARB_shader_group_vote
GL_ARB_shader_image_load_store
GL_ARB_shader_image_size
GL_ARB_shader_objects
GL_ARB_shader_precision
GL_ARB_shader_storage_buffer_object
GL_ARB_shader_subroutine
GL_ARB_shader_texture_image_samples
GL_ARB_shader_texture_lod
GL_ARB_shading_language_100
GL_ARB_shading_language_420pack
GL_ARB_shading_language_include
GL_ARB_shading_language_packing
GL_ARB_shadow
GL_ARB_sparse_buffer
GL_ARB_sparse_texture
GL_ARB_stencil_texturing
GL_ARB_sync
GL_ARB_tessellation_shader
GL_ARB_texture_barrier
GL_ARB_texture_border_clamp
GL_ARB_texture_buffer_object
GL_ARB_texture_buffer_object_rgb32
GL_ARB_texture_buffer_range
GL_ARB_texture_compression
GL_ARB_texture_compression_bptc
GL_ARB_texture_compression_rgtc
GL_ARB_texture_cube_map
GL_ARB_texture_cube_map_array
GL_ARB_texture_env_add
GL_ARB_texture_env_combine
GL_ARB_texture_env_crossbar
GL_ARB_texture_env_dot3
GL_ARB_texture_float
GL_ARB_texture_gather
GL_ARB_texture_mirror_clamp_to_edge
GL_ARB_texture_mirrored_repeat
GL_ARB_texture_multisample
GL_ARB_texture_non_power_of_two
GL_ARB_texture_query_levels
GL_ARB_texture_query_lod
GL_ARB_texture_rectangle
GL_ARB_texture_rg
GL_ARB_texture_rgb10_a2ui
GL_ARB_texture_stencil8
GL_ARB_texture_storage
GL_ARB_texture_storage_multisample
GL_ARB_texture_swizzle
GL_ARB_texture_view
GL_ARB_timer_query
GL_ARB_transform_feedback2
GL_ARB_transform_feedback3
GL_ARB_transform_feedback_instanced
GL_ARB_transform_feedback_overflow_query
GL_ARB_transpose_matrix
GL_ARB_uniform_buffer_object
GL_ARB_vertex_array_bgra
GL_ARB_vertex_array_object
GL_ARB_vertex_attrib_64bit
GL_ARB_vertex_attrib_binding
GL_ARB_vertex_buffer_object
GL_ARB_vertex_program
GL_ARB_vertex_shader
GL_ARB_vertex_type_10f_11f_11f_rev
GL_ARB_vertex_type_2_10_10_10_rev
GL_ARB_viewport_array
GL_ARB_window_pos
GL_ATI_draw_buffers
GL_ATI_texture_float
GL_ATI_texture_mirror_once
GL_EXTX_framebuffer_mixed_formats
GL_EXT_Cg_shader
GL_EXT_abgr
GL_EXT_bgra
GL_EXT_bindable_uniform
GL_EXT_blend_color
GL_EXT_blend_equation_separate
GL_EXT_blend_func_separate
GL_EXT_blend_minmax
GL_EXT_blend_subtract
GL_EXT_compiled_vertex_array
GL_EXT_depth_bounds_test
GL_EXT_direct_state_access
GL_EXT_draw_buffers2
GL_EXT_draw_instanced
GL_EXT_draw_range_elements
GL_EXT_fog_coord
GL_EXT_framebuffer_blit
GL_EXT_framebuffer_multisample
GL_EXT_framebuffer_multisample_blit_scaled
GL_EXT_framebuffer_object
GL_EXT_framebuffer_sRGB
GL_EXT_geometry_shader4
GL_EXT_gpu_program_parameters
GL_EXT_gpu_shader4
GL_EXT_import_sync_object
GL_EXT_multi_draw_arrays
GL_EXT_packed_depth_stencil
GL_EXT_packed_float
GL_EXT_packed_pixels
GL_EXT_pixel_buffer_object
GL_EXT_point_parameters
GL_EXT_polygon_offset_clamp
GL_EXT_provoking_vertex
GL_EXT_rescale_normal
GL_EXT_secondary_color
GL_EXT_separate_shader_objects
GL_EXT_separate_specular_color
GL_EXT_shader_image_load_store
GL_EXT_shader_integer_mix
GL_EXT_shadow_funcs
GL_EXT_stencil_two_side
GL_EXT_stencil_wrap
GL_EXT_texture3D
GL_EXT_texture_array
GL_EXT_texture_buffer_object
GL_EXT_texture_compression_dxt1
GL_EXT_texture_compression_latc
GL_EXT_texture_compression_rgtc
GL_EXT_texture_compression_s3tc
GL_EXT_texture_cube_map
GL_EXT_texture_edge_clamp
GL_EXT_texture_env_add
GL_EXT_texture_env_combine
GL_EXT_texture_env_dot3
GL_EXT_texture_filter_anisotropic
GL_EXT_texture_integer
GL_EXT_texture_lod
GL_EXT_texture_lod_bias
GL_EXT_texture_mirror_clamp
GL_EXT_texture_object
GL_EXT_texture_sRGB
GL_EXT_texture_sRGB_decode
GL_EXT_texture_shared_exponent
GL_EXT_texture_storage
GL_EXT_texture_swizzle
GL_EXT_timer_query
GL_EXT_transform_feedback2
GL_EXT_vertex_array
GL_EXT_vertex_array_bgra
GL_EXT_vertex_attrib_64bit
GL_EXT_window_rectangles
GL_IBM_rasterpos_clip
GL_IBM_texture_mirrored_repeat
GL_KHR_blend_equation_advanced
GL_KHR_context_flush_control
GL_KHR_debug
GL_KHR_no_error
GL_KHR_robust_buffer_access_behavior
GL_KHR_robustness
GL_KTX_buffer_region
GL_NVX_conditional_render
GL_NVX_gpu_memory_info
GL_NVX_multigpu_info
GL_NV_ES1_1_compatibility
GL_NV_ES3_1_compatibility
GL_NV_alpha_to_coverage_dither_control
GL_NV_bindless_multi_draw_indirect
GL_NV_bindless_multi_draw_indirect_count
GL_NV_blend_equation_advanced
GL_NV_blend_square
GL_NV_command_list
GL_NV_compute_program5
GL_NV_conditional_render
GL_NV_copy_depth_to_color
GL_NV_copy_image
GL_NV_depth_buffer_float
GL_NV_depth_clamp
GL_NV_draw_texture
GL_NV_explicit_multisample
GL_NV_fence
GL_NV_float_buffer
GL_NV_fog_distance
GL_NV_fragment_program
GL_NV_fragment_program2
GL_NV_fragment_program_option
GL_NV_framebuffer_multisample_coverage
GL_NV_geometry_shader4
GL_NV_gpu_program4
GL_NV_gpu_program4_1
GL_NV_gpu_program5
GL_NV_gpu_program5_mem_extended
GL_NV_gpu_program_fp64
GL_NV_gpu_shader5
GL_NV_half_float
GL_NV_internalformat_sample_query
GL_NV_light_max_exponent
GL_NV_multisample_coverage
GL_NV_multisample_filter_hint
GL_NV_occlusion_query
GL_NV_packed_depth_stencil
GL_NV_parameter_buffer_object
GL_NV_parameter_buffer_object2
GL_NV_path_rendering
GL_NV_pixel_data_range
GL_NV_point_sprite
GL_NV_primitive_restart
GL_NV_register_combiners
GL_NV_register_combiners2
GL_NV_shader_atomic_counters
GL_NV_shader_atomic_float
GL_NV_shader_buffer_load
GL_NV_shader_storage_buffer_object
GL_NV_shader_thread_group
GL_NV_texgen_reflection
GL_NV_texture_barrier
GL_NV_texture_compression_vtc
GL_NV_texture_env_combine4
GL_NV_texture_multisample
GL_NV_texture_rectangle
GL_NV_texture_shader
GL_NV_texture_shader2
GL_NV_texture_shader3
GL_NV_transform_feedback
GL_NV_transform_feedback2
GL_NV_uniform_buffer_unified_memory
GL_NV_vertex_array_range
GL_NV_vertex_array_range2
GL_NV_vertex_attrib_integer_64bit
GL_NV_vertex_buffer_unified_memory
GL_NV_vertex_program
GL_NV_vertex_program1_1
GL_NV_vertex_program2
GL_NV_vertex_program2_option
GL_NV_vertex_program3
GL_S3_s3tc
GL_SGIS_generate_mipmap
GL_SGIS_texture_lod
GL_SGIX_depth_texture
GL_SGIX_shadow
GL_SUN_slice_accum
GL_WIN_swap_hint
WGL_EXT_swap_control

### Behavior
I lauch Prusaslicer, the Plater tab is white. The build plate rendering is impossible.

[image]

enricoturri1966, 2020-03-16:
@ercoleg
Could you please check if this issue is related to: https://github.com/prusa3d/PrusaSlicer/issues/3587 ?

bubnikv, 2020-03-16:
Also some people reported that it helps to run PrusaSlicer in admin mode, but we don't have a good explanation to it.

Is the rest of the application running? Are the menus and the side panel working for you?

May you please try the other PrusaSlicer 2.2.0 releases and find the first one that hangs? You may try the https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0-alpha2 first.

ercoleg, 2020-03-16:
@enricoturri1966 issue doesn't seem related to #3587 

@bubnikv I have tryed to lauch Prusa Slicer in admin mode with the same result, the rest of the application seems to run, when I come back to plater tab the plater area is not refreshed and keeps the previous tab informations.
I have tried all PrusaSlicer 2.2.0 release since alpha having always the same problem.

I tried also the test SW supplied by @enricoturri1966 in issue #3008 but it has always the same issue

Thank you for your support

bubnikv, 2020-03-16:
>the rest of the application seems to run, when I come back to plater tab the plater area is not refreshed and keeps the previous tab informations.
I have tried all PrusaSlicer 2.2.0 release since alpha having always the same problem.

That means the 3D rendering on plater works initially, you can populate plater with objects, then you slice an switch to 3D path preview and it hangs? 

Would you please make a video for us?

Did PrusaSlicer 2.2.1 work for you?

bubnikv, 2020-03-16:
would updating your graphics card drivers help? Buggy opengl drivers are often the culprit.
nahimic driver as well https://github.com/prusa3d/PrusaSlicer/issues/2522

ercoleg, 2020-03-17:
@bubnikv I have done a video to better explain the issue: [youtube video](https://youtu.be/FM6YWXg7cVA)
I have updated grafic card drivers to latest release (quite old):
NVIDIA NVS 4200M
Provider: NVIDIA
Version: 22.21.13.8205
Date: 05/01/2017
Intel® HD Graphics 3000
Provider: Intel Corporation
Version: 9.17.10.4459
Date: 19/05/2016
in this file all the details
[Detailed-System-Report (1).zip](https://github.com/prusa3d/PrusaSlicer/files/4343281/Detailed-System-Report.1.zip)

bubnikv, 2020-03-17:
Would you please test with https://www.dropbox.com/s/bqq391boywhvsza/PrusaSlicer-2.2.0-rc4%2B12-glsafe-win64-g1ce9ce14b-202003171320.zip?dl=0

Please run prusaslicer-console.exe . This build shall print out errors reported by the OpenGL driver when / if they happen. It may or may not give us some hints of what is happening.

ercoleg, 2020-03-17:
Hi,
I have tested your build:
launched prusaslicer-console.exe as administrator
same behavior showed in the video
I tryed also to open an STL file, I have sliced it but no rendering appears in the build plate tab.

**no error printed in the console during the whole test.**

bubnikv, 2020-03-17:
thanks for testing anyways.

You may try to switch to the integrated Intel graphics card, you may also
try software rendering by running prusa-slicer with  --sw-renderer to
render using an integrated MESA software renderer, though slowly. I am
afraid that that is all we can do for you.

ercoleg, 2020-03-17:
@bubnikv running prusa-slicer with  --sw-renderer solves the issue and I'm fine with this solution.

Thank you for your precious support!
Gabriele

[image]

closed 2020-03-17 by ercoleg

bubnikv, 2020-03-18:
thanks, It is not a great solution, but at least it works for you now.