prusa3d/PrusaSlicer #3846: Incorrect Perimeter slice using modifiers

GWdd, 2020-03-15:
### Version
2.2.0

### Operating system type + version
Win 10 1909

### 3D printer brand / version + firmware version (if known)
Prusa Mk3

### Behavior
Using a cylindrical modifier to increase number of perimeters at holes in a part, the modifer acts more like it is a part, adding perimeters around the modifier rather than the actual hole in the part.

See attached .3mf for an example.  The goal was to increase the hole perimeters to 3, but the modifier is behaving oddly.

The hole should have 3 perimeter instead of the 2 set for the entire part in Printer Settings.
* _Actual Results_
Six or more perimeters are being created...
[image]
[image]

Bug, not Feature

#### Project File (.3MF) where problem occurs
[3MF.zip](https://github.com/prusa3d/PrusaSlicer/files/4335427/3MF.zip)

bubnikv, 2020-03-17:
The way the modifier volumes work is they split the object into pieces, which are sliced independently. Thus if you changed the number of perimeters, this part gets its perimeters, which number is different from the main object configuration.

The way you expect PrusaSlicer to override number of perimeters is significantly more involved to implement than what we currently have.

tcurdt, 2020-06-19:
@bubnikv Thanks for explaining the why. It might be expected as Prusa Slicer developer, but I'd argue it is unexpected from the normal user's POV.

I don't know how the algorithm works - but is there maybe a way to differentiate between an inner and an outer wall? If so the perimeter for outer walls could be set to 0, while the perimeter of inner walls could be set to the desired value.

ctilley83, 2021-01-16:
@bubnikv Instead of separating the original part into pieces using the modifier as a cutter, you could just use logic that if any perimeters are detected within the bounds of the modifier, that those areas only get X amount of perimeters. I see no need to separate the part given that logic already exists where perimeters are to be placed and whether they connect. Seems the wrong approach was applied for this particular type of modifier. I see no use case where the current feature is even remotely satisfactory if it always adds unneeded perimeters.  The layers and perimeter modifier type should be removed from PS until solved. I mean can you think of a use case where the current behavior gives the expected result?

neophyl, 2021-01-16:
I've used it around holes that I want extra perimeters on.  When for example I need to screw to hold in the plastic itself.  I can drop a cylinder over the hole and set a different perimeter value and it then gives me lots.  More than I expected initially but once you know what it does and plan accordingly its not a problem.
So theres one use case for you.

GWdd, 2021-01-16:
I disagree that it can be worked around in all cases. Sure, for simple holes you can - with trial and error - get it to almost do what you want. But that isn't how CAD tools should work.  I've gone so far as to place a modifier so that top and bottom margins are within a part, to keep the extra perimeters it adds hidden for the top surface, but that took a lot more time than is reasonable. These days if one area of a part needs 4 perimeters, I just use 4 perimeters in the first place. Yep - I avoid using this 'feature' because of it's side effects.

Should the 'feature' be removed? Probably not. As Neo said - there are extremely rare circumstances it can still be used.

ctilley83, 2021-01-16:
> I've used it around holes that I want extra perimeters on. When for example I need to screw to hold in the plastic itself. I can drop a cylinder over the hole and set a different perimeter value and [...]

You admit it doesn't work as intended and gives you more perimeters than you expect. So the use case you state is a byproduct of an incorrect implementation.  I get the feature is hard to implement, but you cant really say the current behavior is remotely satisfactory.

ctilley83, 2021-01-27:
Noticed this was closed to reduce duplicate issues.  Has a decision been made to investigate and implement a solution?

bubnikv, 2021-06-03:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/3370

itrmc, 2022-06-14:
Would love to see this capability. When printing a vertical threaded hole, adjusting the number of perimeters can dramatically reduce internal infill and improve time. In the attached image, the model on the left has an inset cylinder modifier meant to increase the number of perimeters in the hole to 4. As you can see in the sliced image, 6 extra perimeters are added at the edge of the modifier.

The second/center object has its perimeters set to 4, eliminating the massive amount of internal infill shown in the unmodified object.

The goal is to have only the threaded hole have 4 perimeters, creating the fastest-printing part.

There is another issue requesting paint-on perimeter modifiers. Perhaps this would be easier to implement.

[image]
[image]

stepikovo, 2022-06-20:
@itrmc A workaround - make the modifier's diameter just slightly bigger than the hole so you will not have the perimeters in the middle of the model, and also make the modifier slightly shorter than the model so it's not crossing the shell on top/bottom, making the top/bottom layers with 2 perimeters and looking normal as the rest of the model.

ctilley83, 2022-06-20:
> @itrmc A workaround - make the modifier's diameter just slightly bigger than the hole so you will not have the perimeters in the middle of the model, and also make the modifier slightly shorter than [...]

Nice tip.  I’ll try this.

ctilley83, 2022-06-20:
> @itrmc A workaround - make the modifier's diameter just slightly bigger than the hole so you will not have the perimeters in the middle of the model, and also make the modifier slightly shorter than [...]

Nice Tip. I’ll try it out.

NathanielArmer, 2025-02-18:
Just ran into this. I expect a modifier to just 'modify' not create brand new geometry (though someone might find that useful).

Use case: printing a vase with a thicker interior for waterproofness:

* 19 hours @ 2 layers.
* 27 hours @ 4 layers all over.
* 20.5 hours @ 4 layers just on the interior (using a carefully positioned modifier that kinda works).

snilard, 2026-02-10:
6 years later and it is still open 😢

MonoTovarisj, 2026-09-18:
still open and the bot is trying to shut it down, maybe nobody cares enough.