prusa3d/PrusaSlicer #3848: Z positioning wrong after tool switch

VanessaE, 2020-03-16:
### Version

2.2.0-rc3

### Operating system type + version

Debian sid

### 3D printer brand / version + firmware version (if known)

Hypercube with two extruders feeding one hotend via a splitter.

### Behavior

When Prusaslicer issues a tool change via T0, T1, ... Marlin automatically recalls the last Z position for the newly-selected tool.  Pretty sure that's firmware intended behavior.

However, Prusaslicer is not issuing any kind of G1 Znnnn command to counteract this, so on the first tool switch to T1, some odd position is engaged, like say 2.55 when Z is supposed to be 0.2. 

I don't know if it affects other layers, but logic tells me it'll happen on the first tool switch even if that switch happens several centimeters up in the print.

A workaround is to put this little stub at the end of the tool change g-code:

```
; force Z to move to the proper height after toolchange
T[next_extruder]
G1 Z[layer_z]
```

See also #1339 and related issues, which refer to temperature being improperly-set on toolchange.

#### Project File (.3MF) where problem occurs

N/A

bubnikv, 2020-03-16:
> When Prusaslicer issues a tool change via T0, T1, ... Marlin automatically recalls the last Z position for the newly-selected tool. Pretty sure that's firmware intended behavior.

What Marlin? Which version?

How is the behavior of the upstream slic3r in this regard? Do you know how other slicers handle this? How about Cura?

We tested on Sigma and I don't think this was an issue.

I don't want to pick on Marlin guys too much, they are doing great job, but sometimes they do a breaking change and we suffer. See for example https://github.com/prusa3d/PrusaSlicer/issues/1089#issuecomment-410262095

VanessaE, 2020-03-16:
> What Marlin? Which version?

My copy of Marlin is the `bugfix-2.0.x` branch, git commit https://github.com/MarlinFirmware/Marlin/commit/484e1a624c178f1bb0996892d2c606f6cfebefbe

> How is the behavior of the upstream slic3r in this regard? Do you know how other slicers handle this? How about Cura?

I have not tried upstream -- it only has rudimentary multi-material support, not even a purge tower.  I don't know about Cura's support here.

> We tested on Sigma and I don't think this was an issue.

What is Sigma?

bubnikv, 2020-03-16:
> What is Sigma?

Don't remember, some 3 or 4 year old IDEX (independent dual exruder) setup.
Got knows what firmware runs on it and we are home bound due to COVID19.

VanessaE, 2020-03-16:
> home bound due to COVID19

Yeah, @josefprusa mentioned it on Twitted.  That sucks :confused:  

But there's a bright side:  lots more time to code :grin:

RcTomcat1, 2020-10-15:
i can confirm this.
I am running a self build printer with a E3D Chimera+ Hotend.
During toolchange the z axis moves up and seems to "forget" the actual z height that it is supposed to print at. The second extruder prints into thin air.

RcTomcat1, 2020-12-02:
Issue still persists with the latest beta

traderhut, 2022-03-24:
Any progress on this in the last 15 months?  I just confirmed the issue with E3D's Tool Changer, it switches tools and leaves it at Z+5 (or so) from where it was, until the next layer, it air-prints all over the print, then on the next layer, it moves down where it should be, switches tools and dumps all over the print again...

Trying to figure out a work around...  Obviously, only an issue if you have multiple tools..

traderhut, 2022-03-24:
(I'm not using the wipe tower, but maybe that would fix it?  As it would force it to move to the wipe location, and it would have to move back...)  I plan on using a wipe tower soon.

RcTomcat1, 2022-03-24:
@traderhut i did use a wipe tower when i initially commented in this issue thread. Didn't work for me, despite using the tower my printhead would print into thin air upon tool change.
This problem is the reason why I had to abandon PrusaSlicer.
What good is a great slicer when it can't be used to print something with my machines?
Please let us know if a wipe tower fixes it for you now. That might convince me to give it another try

traderhut, 2022-03-24:
@RcTomcat1   I might be able to do you one better... I might just modify the slicer and do a pull request back for them.  It simply needs to do a G1 Znn  command after the Tnn command in the gcode.  The best fix is to have another edit box for 'post tool change' that would allow you to enter whatever commands you wanted to do there...     I really like the slicer as well, but clearly, it should either always set the z after doing a tool change, or give you an option to run code following a tool change.  (They actually make a comment on that page in the settings that you can do that, but I don't see any way to actually do it..

I've not tried to compile the software on my machine yet, so that could end up being an issue.

traderhut, 2022-03-24:
Oh, there is a solution I came up with, it seems crazy, but it would work.  Following a T1-T3, have the last thing it does is set it relative, and then G1 Z-nnn    Where nnn is taken from the difference between the heights of the nozzles.  (in my case 1.6mm), this would be replacing the code that lowers the bed 5mm.   I've not tried this yet (I did remove the lowering of the bed, and because my T1 tool sticks down farther, it crashed into the bed, but that is because I didn't do a G1 Z1.6  to get it back up to the right level.   The down side of this is that when you change nozzles, you have to not only update the z axis offsets in config.g or config-overide.g  but also modify the tool change code.

traderhut, 2023-04-28:
Been a year without being able to use the current branch of the software (I'm currently trying to merge my changes into the current branch, but my changes for this is a hack that saves the Z off, sets a flag when the tool was changed and on the next move, if the tool was changes, issues a Z command that matches the last Z - in order to fix the +5mm difference on the Z position following the tool change.  It then moves to where it should be and continues the print.  This works well, but I'd like to pick up the latest changes and be able to get some of the new features without having it air-print after each tool change.

The other change that I have to do to print multi-tool is to create my 'floating' 2nd tool model so it has a 0.01mm line going down to the 1st layer, off to a small circle of material that is 0.2mm high (outside of the normal print area), this keeps the software from blowing up with the error that that part has nothing on the 1st layer.. it still issues a warning that nothing exists between 0.2mm and wherever the part is - but that is another defect/issue.  (If I can paint on top of a part a different tool/color, and that works - floating up there, clearly the slicer can support my doing that intentionally!

Anyway, would sure be cool if this ticket got fixed so I could stop patching and rebuilding the code...

alexjx, 2023-06-22:
I think the problem is that slicer doesn't generate Z coodinate after tool change. It's assuming the tool swich is returning the exact offfset especially the Z offset after the tool has switched.

I'm starting working with a E3D tool changer, and suprised by this when I switched from Cura to PS.

Although I could use `G1 R2` to restore the remembered position, this is very bad for the print. Since most of the the time the start position of next tool normally not the end position of the previous one.

for example in the following picture T0 finishes at the square marker, and T1 start at the circle. And the only thing I could do is to restore T1 to the square, and continue the move from there to the circle.

While cura will generate a z offset in the first G0/G1 command right after the tool change. This will save us from doing the position restore, and just go to the start position after tool switch.

The extra Z offset will just be a no-op for a single extruder multi material case. So there should be no harm in this case. 

Jia

traderhut, 2023-06-24:
I thought about the Save/Restore Tool Position, but it doesn't work.   The reason is that let's say that you have offsets for Tool 0 at (0,0,0) (there are values in the system for X,Y,Z that defines where the tool tip is), and Tool 1 is (1,1,1). This would mean that the Tool 1's location of (0,0,0) the Toolchanger would actually position it to 1,1,1 to correct for the tool offset.

If you do this: (PsudoCode)

T0     -- At some point
<Save position>.    (Let's say you are at 2,2,2, with offsets also 2,2,2)
 (Move +5 from where you are)
T1.   - Which switches to offsets of 1,1,1
<Restore Position>.    Returns to absolute position of 2,2,2, which is actually position 1,1,1!! Do to offsets!

If you then tell it to move to 2,2,2.  it will move to 3,3,3

I had so much promise for this, but alas, the firmware that implements the save/restore position is using the position after offsets are applied, and restores to that same position, and the saved positions are not updated by a change in the offsets.  (I consider this a bug in the firmware actually, but if it was corrected, it would break existing code - I assume)

Anyway, my solution was to download the code, build it, and then hack it (Horrible hack, IMHO as a software Architect - Any time you move, I save where you moved to, and after a Tool change, I simply go there.   The next move is pretty much always a G0 to just X and Y.  Actually, I think I do a G0 to the correct X and Y of where you were, and then do a G0 to set the Z, this keeps it from swooping down across the part and maybe hitting something that already got printed on that layer.

I would have just used the current X, Y and Z in the program, but couldn't for the life of me quickly figure out where that was, but I was able to find a single function that handled all the moves, so saved the location in a place that I could access later.  (Like I said, this was a quick hack to make it work)

I've printed a LOT of multi color prints after that..

Oh, and it is kind of strange too, because if your offsets were high and you just restored you could have the head smash into the print (Actually, it would alternate between smashing into it and air printing - I got lucky and it air printed, which took me a while to figure out, but for some reason my Z was off by about 0.8mm or so..)

I'm pretty sure that you either have to manually edit the gcode file (or write a program to do so), or you have to modify the slicer to fix the problem.

traderhut, 2023-06-24:
Btw: It should generate the XY move FIRST, and THEN move the Z down to the right place.  And yes, it is best to move to the NEW location and not the old one, as this is a perfect situation where the Z is already +5mm (in my case) above the print so moving to the right location won't bump the print

alexjx, 2023-06-25:
@traderhut I tried my idea of generating Z offset after each tool change. This is the behavior of Cura. And it works as I expected.

[image]

The idea is simple, 
1. I add a flag in gcode writer.
2. the flag will be set once the tool changed.
3. the z coordinate will be generated in the next g0/g1 move. and reset the flag.

The z coordinate is inserted AFTER XY move, as you mentioned. 

This will establish the correct z coordinate after tool change. So it doesn't rely on the firmware to restore the `last position`. We could just do a tool switch and move to next intended print position.

I'm still testing the change, and like you mentioned, there are several places that in prusaslicer will generate move/extrusion instruction. There is not a simple place to do the change.

Jia

pillopaolo, 2023-12-24:
Hi there,
"Retraction Lift Z" will cause Z to be enforced right after the tool change. I actually do not need/want a real Z lift, so I set a very small number, e.g. 0.001, just to get the above problem solved.
If you do not want retration, then set a very small number for "Retraciton Length" too.

...
;WIPE_END
G1 Z1.401 F9000
T1
G1 X218.417 Y201.769
G1 Z1.4
G1 E.3 F6000
;TYPE:Support material

jwalker55, 2024-03-03:
I am now suddenly having this problem after updating from PrusaSlicer 2.6 to 2.7.2. After a tool change, it now does not restore the Z position and just prints in mid air. @pillopaolo mentioned "Retraction Lift Z" but I don't see such a setting anywhere?

pillopaolo, 2024-03-03:
Hi, it is in the Retraction menu, after Length. Worse case use Search to
find it.

jwalker55, 2024-03-03:
> Hi, it is in the Retraction menu, after Length. Worse case use Search to find it.
> […](#)

This setting does not exist in PrusaSlicer 2.7.2

closed 2026-09-29 by prusaslicer-bot[bot]