prusa3d/PrusaSlicer #3851: fetching PrusaSlicer-settings-master assumes wrong ca path on some linux

draeath, 2020-03-16:
### Version
2.2.0-rc4+linux+x64 (`PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.AppImage`)

### Operating system type + version
Linux x86_64 5.5.7-1-default (openSUSE tumbleweed 20200312)

### 3D printer brand / version + firmware version (if known)
i3 MK3S

### Behavior
The following messages are observed in the terminal when launching PrusaSlicer:

```
[2020-03-16 13:43:56.679744] [0x00007f7df11bd700] [error]   Error getting: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]
[2020-03-16 13:43:57.108467] [0x00007f7df11bd700] [error]   Error getting: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx`: HTTP 0, Problem with the SSL CA cert (path? access rights?):
error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
[Error 77]
```

On my distribution, the file /etc/ssl/certs/ca-certificates.crt does not exist, contrary to other rpm-based distros. I'm also unsure how debian/ubuntu place the file.

Consider also including checks for `/etc/ssl/ca-bundle.pem` and perhaps reviewing the way this is implemented - the host's openssl/gnutls should be handling this for you shouldn't it?

I am able to work around this issue by placing a symbolic link such that PrusaSlicer's attempts to access the "wrong" filename are successful.

I am unsure if update checking is actually broken when this isn't done.

(just a note, when tested with CURL, the URLs work (caveat: the http URL responds with a 301)

```
bransfp@ginnungagap:~> curl -I http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch//index.idx
HTTP/1.1 301 Moved Permanently
Server: nginx
Date: Mon, 16 Mar 2020 17:44:09 GMT
Content-Type: text/html; charset=iso-8859-1
Connection: keep-alive
Location: https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaResearch/index.idx

bransfp@ginnungagap:~> curl -I https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version
HTTP/2 200 
server: nginx
date: Mon, 16 Mar 2020 17:44:19 GMT
content-length: 6
last-modified: Mon, 16 Mar 2020 17:05:15 GMT
etag: "6-5a0fbd2b24c3b"
accept-ranges: bytes
```

#### Project File (.3MF) where problem occurs
N/A

bubnikv, 2020-03-17:
Thanks for heads up.

We compile libcurl on old centos version docker image using latest devtools to produce binary builds against old glibc. I suppose the libcurl just configures its paths to certificates according to centos rules.

Unfortunately we did not think about those distro incompatibilities an I am afraid we would have to patch the libcurl configuration process or possibly libcurl.

Without the certificates, likely the automatic update of print profiles does not work.

bubnikv, 2020-03-17:
/etc/ssl/certs/ca-certificates.crt
exists on Debian Jessie and Ubuntu 16.
at least that seems to work.

commit 1123689a (2020-03-25): Search for openssl cert store on app init, if the store is not present.

fixes  #3851
closed 2020-03-25 by tamasmeszaros