prusa3d/PrusaSlicer #3852: 3DConnexion SpaceMouse settings not working.

Gasol1n, 2020-03-16:
### Version
2.2.0-rc4+win64

### Operating system type + version
Windows 10 (build 18362), 64-bit edition

### Behavior
I prefer to zoom with Up and Down movements on my SpaceMouse, rather then forward and back. But when i change the Zoom direction setting, it stays the same.
Also every other settings option seems not to work as well. I can't invert and/or disable axis.

bubnikv, 2020-03-16:
PrusaSlicer does not communicate with the official driver on Windows,
therefore you cannot reconfigure axes in the driver.

We received an extremely poor support from 3DConnexion, they provided us
with the drivers after half a year,therefore we had to find a way.

jurcja1, 2020-03-17:
In the meantime when PrusaSlicer does not communicate with official driver, what is really poor, please add simple menu where will be possible to set behaviour of 3D mouse as each user wants. Then you can develop interface between official 3DConnexion driver and PrusaSlicer, but implementation of 3D mouse will be working in adequate way.

[image]

jweaver100, 2020-03-19:
CTL+M

That brings up the menu where you can adjust sensitivities.. But you cant' remap/change the functions like you do with the normal drivers.. Luckily for me, I use mine exactly how Slic3r has implimented them, so i am quite fortunate.

foreachthing, 2020-03-19:
@jweaver100 , yeah! Same here!
But I guess a "reverse axis" checkbox should not be that hard to implement (if you know your way around c++) and would help others.

bubnikv, 2020-03-20:
Would you guys please share screenshots of the 3Dconnexion driver dialog documenting how your prefer the axes of the 3DConnexion device to be shuffled? I don't think we need to implement the full freedom of the 3DConnexion configuration dialog.

jurcja1, 2020-03-20:
@bubnikv Yes, I shared basic settings, but I cannot test it due to non function so It can be different according your implementation. In addition, I do not think that one settings will cover all request, habits and needs of designers, at all.

bubnikv, 2020-03-20:
> In addition, I do not think that one settings will cover all request, habits and needs of designers, at all.

Frankly I don't understand the full configurability of the 3d mouse. It does not make sense to me for example to flip an X axis, so you will press right and the object will move left. In the air plane RC control world there is the A and B configuration, more does not make sense.

Though I do not underestimate creativity of the users if you give them the knobs to wiggle with.

jurcja1, 2020-03-20:
Basically, it needs to understood possibilites of 3D mouse at all. It is not only about moving with object but especially shortcuts but mouse buttons etc. If there will not be direct connection with original driver, what is possible to do, it will never work fully as it should. I use a lot of environments and trust me, I have differencies everywhere, especially 3D environment.

bubnikv, 2020-03-20:
@jurcja1 You can still map the keys using the driver.

jurcja1, 2020-03-20:
@bubnikv Yes, but completely manually. Checking shortcut in Slicer, setting it in 3Dconnexion driver. Not directly as function mentioned in 3Dconnexion driver, same as it works with other design software. When shortcut will be changed in Slicer, 3Dconnexion driver holds shorcut instead of function. From my point of view, better is implement 3Dconnexion driver instead of developing own it itself and tuning all errors. Do not know Prusa relationship with 3Dconnexion company, but I think they are able to provide you source documentation and implementation docs for Slicer platform briefly, as another reference of design software for their products.

Kachidoki2807, 2020-03-20:
Axis flipping is useful to change the "point of view". I mean, at the very beginning, I was using the 3D mouse as a camera controller. So if I press right, the camera goes to the right, and therefore the object goes to the left.

After some times, looking for the best comfort settings, I switched to a very simple configuration. Let me try to explain it:
If we think in term of planes references, your 2D mouse goes on X and Y but the plane reference is your desk. But physically, your mouse cursor will go on a plane reference that is "vertical", your screen. That mean that 2D XY plane become technically a 3D XZ plane.

So the best comfort settings for me is to handle the 3D mouse like if I handle the object with my hand, but with a 90° flip, like the 2D mouse. So to achieve that, the OP's suggestion to zoom by Up/Down (on Z axis, desk as reference), is the best natural way.

Hoping to be clear.

bubnikv, 2020-03-20:
> So the best comfort settings for me is to handle the 3D mouse like if I handle the object with my hand, but with a 90° flip, like the 2D mouse. So to achieve that, the OP's suggestion to zoom by Up/Down (on Z axis, desk as reference), is the best natural way.

We have implemented that option for the final release. I suppose this covers most of the complains
Closing.

closed 2020-03-20 by bubnikv

GarthSnyder, 2020-03-20:
Without the ability to configure axes, I'm afraid that for me PrusaSlicer's current SpaceNavigator support has no utility.

This isn't really a "user preference" issue. Nobody's buying a SpaceNavigator just to use it in PrusaSlicer. We own them because we use them in other 3D applications, which vary in their configuration. It just isn't possible to switch back and forth between different configurations; it's as if whenever PrusaSlicer was the active app, moving the mouse up moved the mouse cursor down. You'd never get used to it.

bubnikv, 2020-03-21:
> This isn't really a "user preference" issue. Nobody's buying a SpaceNavigator just to use it in PrusaSlicer. We own them because we use them in other 3D applications, which vary in their [...]

I understand your stance.

Now please understand ours.

We cannot give everybody every knob to wiggle we can think of. We have to find out the real world use cases first to find the lowest common denominator fitting the majority. Giving people all the knobs they can imagine leads to an unmaintainable mess.

Now people usually have hard time to explain what they want. Likely the most effective process in computer user interface design is to feed the users prototypes with increasing numbers of knobs to give back their feedback so that we will find the least common denominator together. The issue I see from my previous experience is once you give somebody a knob, they will scream terribly if you take that knob away from them, even if the knob has questionable utility.

Gasol1n, 2020-03-21:
Thank you for implementing a Zoom Direction Switch.
That was the only thing i have changed from the default settings on my 3D Mouse and i think most of the people are also satisfied with this.

Also, a normal Mouse still works the same if someone is not happy with the SpaceMouse control Sceme.

Kachidoki2807, 2020-03-22:
I have to point out that the Release 2.2.0 with the option to swap Y/Z axes is a pure delight!
Remember guys, you can find it out by pressing Ctrl+M.

Thank you @bubnikv and @enricoturri1966

bubnikv, 2020-03-22:
>  I have to point out that the Release 2.2.0 with the option to swap Y/Z
axes is a pure delight!

Great, thanks for heads up.

Now I am just waiting for those that configured their 3D mouse using the
system driver yet another way :-)

bringbackgroupb, 2020-09-03:
> We cannot give everybody every knob to wiggle we can think of. We have to find out the real world use cases first to find the lowest common denominator fitting the majority. Giving people all the [...]

I really don't think that asking for this change is asking for "too many knobs". It's really a required base functionality for 3d mice. To address your 2nd topic, hopefully I can clearly communicate the issue. There are two ways of thinking when it comes to 3d mouse use:

1.  The mouse acts as a proxy for the object and by moving the mouse you are moving the object within the viewport (currently implemented)
2. The mouse acts as a proxy for the camera and it allows you to move the camera around the viewport (my perference)

As mentioned before, no one is buying a 3dconnexion mouse to use it in Prusa Slicer alone, we are usually using them in other CAD packages. Often, we have been using a 3d mouse for decades to navigate our workspace so changing our mindset for a single piece of software is not only difficult, it's probably a bad idea (leads to mental confusion).

An example for this would be controller mapping for video games. Some people prefer the "up is up" configuration while others prefer the "flight sim" configuration where you pull up and push down. No one is arguing one of these two options is invalid, they're just personal preference an often something someone can't (or has no desire) really relearn.

Please reconsider and include the functionality to reverse the axis directions for these mice, I love Prusa Slicer (I created a github account to post this message) but it's very frustrating to have a feature you've always wanted delivered just to find that it won't work for you.

bubnikv, 2020-09-03:
For PrusaSlicer 2.3 we will use a channel that goes through the 3DConnexion
driver, so you will be able to configure the buttons in a usual way.
3DConnexion is supposed to put our configuration file into their driver,
keep your fingers crossed.

GarthSnyder, 2020-09-03:
That's great news, @bubnikv. I'm happy that 3DConnexion eventually stepped up to the plate.

Even so, the original implementation of SpaceMouse support was quite good. This is a difficult peripheral to support correctly, and quite a few implementations end up making a hash of it. I was one of the earlier complainers about the lack of axis swapping, but the minimal support that PrusaSlicer eventually included solved the issue for me. I've been using my SpaceMouse quite happily in PS ever since.