prusa3d/PrusaSlicer #3854: 2.2.0-RC4 vendor profile issue "missing" / filament used for two vendor profiles

photodude, 2020-03-17:
### Version
Build:     PrusaSlicer-2.2.0-rc4+win64-202003151137

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 18363), 64-bit edition
Total RAM size [MB]: 34,307MB
OpenGL installation
GL version:   4.6.0 NVIDIA 441.87
Vendor:       NVIDIA Corporation
Renderer:     GeForce GTX 1070/PCIe/SSE2
GLSL version: 4.60 NVIDIA

### 3D printer brand / version + firmware version (if known)
BIBO2 Stock model with 5015 part cooling blower and PEI sheet build surface
BIBO MFG provided stock based on [Marlin 1.0.0 shaqFoo Ditto printing fork](https://github.com/ShaqFoo/Marlin_Firmware_Deluxe_With_Ditto_Print-Auto_Leveling) firmware

### Behavior
Opened 2.2.0-RC4 selected "add new printer" take snapshot/remove user profiles finish add dialog for BIBO printer and get error about missing profile. 

* _Steps needed to reproduce the problem_

1. Opened 2.2.0-RC4
2. printer menu
3. selected "add new printer"
4. take snapshot/remove user profiles
5. finish add dialog for BIBO printer
 and get error about missing profile or filaments used by another profile. 

first attempt showed two BIBO vendors
cleared all files from the configuration folder `C:\Users\[user]\AppData\Roaming\PrusaSlicer`
new error about missing profile

* _Expected Results_
no errors
* _Actual Results_
unable to re-add vendor profile with RC4 due to vendor profile related errors

* Screenshot
first attempt showed two bibo vendor profiles resulted in the first two errors
[image]
[image]
after clearing configuration folder new error
starts the same as image 1 then this dialog
[image]

* Additional information
unable to find a reason why it's looking for the old vendor profile file name

photodude, 2020-03-17:
Looks like it's an issue with the `C:\Users\[user]\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` file

Specifically in the last section
```ini
[vendor:BIBO2Touch]
model:BIBO2 = 0.4
```
this should be overwritten with the following change as per the updated naming in the vendor profile
```ini
[vendor:BIBO]
model:BIBO2 = 0.4
```

and the updated vendor file should be placed in the 
`C:\Users\[user]\AppData\Roaming\PrusaSlicer\vendor` folder

bubnikv, 2020-03-17:
Are you able to reproduce this situation if you delete your old PrusaSlicer configuration directory and never install any PrusaSlicer with the old vendor:BIBO2Touch definition?

It looks to me that your current state is caused by mixing these alpha profiles of yours with what we currently have in PrusaSlicer 2.2.0-rc4.

bubnikv, 2020-03-17:
I was not sure whether I understood your mention of snapshots, so I also tested reverting back to a snapshot taken before the wizard and then activating the snapshot with the BIBO printer installed and everything seems to be working.

It seems like this issue is really just a result of mixing the alpha BIBO profiles that hopefully nobody installed with the latest profiles. I don't think it is worth fixing.

photodude, 2020-03-18:
@bubnikv I did try deleting all files but the snapshots including `C:\Users\[user]\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` file from the configuration directory. The `C:\Users\[user]\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` file recreated itself. the only workaround was to edit the `C:\Users\[user]\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini` file 

Note: I still have 2.1.0 still installed and it technically still has an old alpha/beta configuration installed although I have not opened it since starting testing with RC3/RC4

A solution is in my comment above if this is decided as something that can't be reproduced, or can't be fixed.

bubnikv, 2020-03-18:
> The C:\Users\[user]\AppData\Roaming\PrusaSlicer\PrusaSlicer.ini file recreated itself.

It recreated itself with the same data only if you had PrusaSlicer running while you deleted the file. 

Again, I hope this is not a real issue as the normal user should not get the names of the installed system profiles messed up.

photodude, 2020-03-21:
@bubnikv we'll take this as a non-reproducible issue with an expected cause and instructions on how to correct the files if removing them doesn't fix it.

Closing as per above.

closed 2020-03-21 by photodude