prusa3d/PrusaSlicer #3856: Add option to exchange stl in object list (right panel)

apphead, 2020-03-17:
Vaguely related to issue#3588

As of PrusaSlicer version 2.2.0-RC4, there is only the option to reload the same stl from disk. It would be handy to be able to exchange an stl for another one and yet keep all the modifiers etc in place. I usually save multiple versions of my stl, so right now, I need to workaround that issue in the filesystem (or start a new plater from scratch)

_Is this a new feature request?_
Yes.

Solution:
Add option below _Reload from disk_ labelled "Replace with stl", which opens a file selecter and thats all ;-)

[image]

bubnikv, 2020-03-18:
if the file is missing, PrusaSlicer will allow you to reload a different
one.

apphead, 2020-03-19:
@bubnikv I tried your workaround, but it failed (2.2.0-RC4):
[image]

In the image above, V12.stl is the initial version, I tried to replace it with V13.stl. Only solution was to rename new file to V12.stl.

bubnikv, 2020-03-19:
I will discuss with @enricoturri1966 who is the author of the feature.

Lukrative525, 2020-03-22:
@apphead I agree this would be super useful, especially when you're iterating on a design.

bcwadell, 2020-04-08:
agree, this is an important feature.  It would have helped when iterating CAD versions of COVID-19 PPE without changing settings that I'd optimized for speed

Mauronic, 2020-04-11:
@bubnikv Did you ever have a conversation with @enricoturri1966 about this?

enricoturri1966, 2020-04-12:
@Mauronic
Yes, we discussed it and planned to allow to replace objects are required. Just waiting to find the time to implement it.

bubnikv, 2021-08-11:
Implemented in PrusaSlicer 2.4.0-alpha1.

closed 2021-08-11 by bubnikv