prusa3d/PrusaSlicer #3857: Error When Trying to Connect to Octoprint Via PrusaSlicer on Linux

chemputer, 2020-03-18:
### Version
2.2.0rc4, as well as 2.1.1, both have the issue. I've also tried both the .appimage versions, and the .tar.gz versions. 

### Operating system type + version
POP OS! 19.10 (based on Ubuntu 19.10) uname -a output:` Linux (hostname) 5.3.0-7629-generic #31~1581628825~19.10~f90b7d5-Ubuntu SMP Fri Feb 14 19:56:45 UTC  x86_64 x86_64 x86_64 GNU/Linux`

### Behavior
* _Describe the problem_
I am attempting to connect to my octoprint instance via the built-in mechanism. It fails, saying the following in a pop-up window whenever I click test. This works aboslutely fine on my Windows 10 machine, running the same PrusaSlicer version, with the same OctoPrint instance, same hostname, and API key. They're both on the same network.

I've tried adding CA files, varying from Let's Encrypt's own, to the root certificate that my own browser seems to be accepting just fine, but nothing works. I suspect that it is due to the haproxy?
```
Could not connect to OctoPrint: Peer certificate cannot be authenticated with given CA certificates:
SSL certificate problem: unable to get local issuer certificate
[Error 60]
Note: OctoPrint version at least 1.1.0 is required.
```
* _Steps needed to reproduce the problem_
Set up an octoprint version 1.4.0 instance, set it up behind haproxy, setup with a Let's Encrypt wildcard cert for one of my domains. I followed [this guide](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337/) to setup octoprint and haproxy, and then [this guide](https://community.octoprint.org/t/restrict-access-to-octopi-utilizing-ssl-client-certificates/3681) to enable SSL, but instead of a self-signed cert, I used a Let's Encrypt Wildcard Cert. This is all running on an arm based Rock64. Try connecting PrusaSlicer on Linux to it.

* _Expected Results_
For it to connect.
* _Actual Results_
It fails to connect, giving the error mentioned above, and the cmd line output below when using --loglevel 4.

_Is this a new feature request?_ No.

```
[2020-03-17 20:53:34.961254] [0x00007fba7440d540] [info]    OctoPrint: Get version at: http://octo.turnerb.com/api/version
[2020-03-17 20:53:34.981864] [0x00007fba7440d540] [error]   OctoPrint: Error getting version: Peer certificate cannot be authenticated with given CA certificates:
SSL certificate problem: unable to get local issuer certificate
[Error 60], HTTP 0, body: ``
[2020-03-17 20:54:14.104138] [0x00007fba7440d540] [info]    OctoPrint: Get version at: https://octo.turnerb.com/api/version
[2020-03-17 20:54:14.116921] [0x00007fba7440d540] [error]   OctoPrint: Error getting version: Peer certificate cannot be authenticated with given CA certificates:
SSL certificate problem: unable to get local issuer certificate
[Error 60], HTTP 0, body: ``
```

bubnikv, 2020-03-19:
Does this help?

https://github.com/prusa3d/PrusaSlicer/issues/2018#issuecomment-601162181

chemputer, 2020-03-19:
Well, I haven't tried it, but no. I don't want to use a self-signed certificate as I have a working, Let's Encrypt Certificate. I know that, as it works on every other place I use it (web servers, the octoprint server), and browsers and apps, and even the Windows version of PrusaSlicer recognizes it fine, and works. If you aren't familiar with those, they're certificates that are free, but that are signed by LE and countersigned by two other major CAs. 

PrusaSlicer should only need a CA file for self-signed SSL certs. Mine isn't self-signed. I'm not connecting via hostname or IP, I'm connecting via FQDN, with a the hostname being a subdomain of one of my domains, and DNS is set up appropriately. 

This is an issue with PrusaSlicer on Linux, not my setup. I wish it weren't.

bubnikv, 2020-03-19:
Could it be that your issue is the same as https://github.com/prusa3d/PrusaSlicer/issues/3851 ?
That is, our build searches system certificates where they are NOT installed on your system?

chemputer, 2020-03-19:
That's certainly possible. Is it specifically looking for /etc/ssl/certs/ca-certificates.crt?

If so, then yes, that could be the issue. In the /etc/ssl/certs folder, there are loads of .pem files, for all the root CAs. 

Is there a way to get the ca-certificates.crt that it's looking for and just put it there? 

The only thing that makes me worry about that not being the problem is that even when specifying a CA file that I know will work with my cert in PrusaSlicer, it gives the same error.

chemputer, 2020-03-19:
After further investigation, I do not believe it is similar to issue #3851. Here's the full output of PrusaSlicer from me opening it from the terminal, with --loglevel 5, going straight to printer settings, and hitting the test button on the octoprint setup. It looks, to me, like it's having issues with haproxy?

```
 % /opt/PrusaSlicer/prusa-slicer --loglevel 5 
[2020-03-19 12:30:43.224361] [0x00007f98c7da7540] [info]    Checking if indices need to be installed from resources...
[2020-03-19 12:30:43.276229] [0x00007f98c7da7540] [trace]   System language detected (user locales and such): en_US
[2020-03-19 12:30:43.276251] [0x00007f98c7da7540] [trace]   Switching wxLocales to en_US
[2020-03-19 12:30:43.657367] [0x00007f98c7da7540] [info]    Undo / Redo snapshot taken: New Project, Undo / Redo stack memory: 0MB Resident memory: 59MB; Shared memory: 41MB; Private memory: 17MB; Peak memory usage: 59MB
[2020-03-19 12:30:45.979216] [0x00007f98c7da7540] [info]    Checking for cached configuration updates...
[2020-03-19 12:30:45.979321] [0x00007f98c7da7540] [info]    Confing bundle not installed for vendor BIBO2Touch, skipping: 
[2020-03-19 12:30:45.979450] [0x00007f98c7da7540] [info]    Confing bundle not installed for vendor PrusaResearch, skipping: 
[2020-03-19 12:30:45.979518] [0x00007f98c7da7540] [info]    Confing bundle not installed for vendor BIBO, skipping: 
[2020-03-19 12:30:45.979557] [0x00007f98c7da7540] [info]    Confing bundle not installed for vendor LulzBot, skipping: 
[2020-03-19 12:30:45.980481] [0x00007f98c7da7540] [debug]   Vendor: Creality, version installed: 0.0.2, version cached: 0.0.2
[2020-03-19 12:30:45.980507] [0x00007f98c7da7540] [info]    No configuration updates available.
[2020-03-19 12:30:46.001038] [0x00007f98c4e18700] [info]    Downloading PrusaSlicer online version from: `https://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version`
*   Trying 185.115.1.124...
* TCP_NODELAY set
* Connected to files.prusa3d.com (185.115.1.124) port 443 (#0)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
* SSL connection using TLSv1.2 / ECDHE-RSA-AES256-GCM-SHA384
* ALPN, server accepted to use http/1.1
* Server certificate:
*  subject: CN=files.prusa3d.com
*  start date: Mar 17 08:24:00 2020 GMT
*  expire date: Jun 15 08:24:00 2020 GMT
*  subjectAltName: host "files.prusa3d.com" matched cert's "files.prusa3d.com"
*  issuer: C=US; O=Let's Encrypt; CN=Let's Encrypt Authority X3
*  SSL certificate verify ok.
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/PrusaSlicer.version HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.2.0-rc4+linux-x64
Accept: */*

< HTTP/1.1 200 OK
< Server: nginx
< Date: Thu, 19 Mar 2020 16:30:47 GMT
< Content-Length: 6
< Connection: keep-alive
< Last-Modified: Thu, 19 Mar 2020 16:16:21 GMT
< ETag: "6-5a1377d53d265"
< Accept-Ranges: bytes
< 
* Connection #0 to host files.prusa3d.com left intact
[2020-03-19 12:30:48.012612] [0x00007f98c4e18700] [info]    Got PrusaSlicer online version: `2.1.1`. Sending to GUI thread...
[2020-03-19 12:30:48.013842] [0x00007f98c4e18700] [info]    Syncing configuration cache
[2020-03-19 12:30:48.013864] [0x00007f98c4e18700] [warning] No such vendor: BIBO2Touch
[2020-03-19 12:30:48.013872] [0x00007f98c4e18700] [warning] No such vendor: PrusaResearch
[2020-03-19 12:30:48.013880] [0x00007f98c4e18700] [warning] No such vendor: BIBO
[2020-03-19 12:30:48.013887] [0x00007f98c4e18700] [warning] No such vendor: LulzBot
[2020-03-19 12:30:48.013905] [0x00007f98c4e18700] [info]    Downloading index for vendor: Creality
[2020-03-19 12:30:48.013940] [0x00007f98c4e18700] [info]    Get: `http://files.prusa3d.com/wp-content/uploads/repository/PrusaSlicer-settings-master/live/Creality//index.idx`
	-> `/home/turnerb/.PrusaSlicer/cache/Creality.idx-update`
	via tmp path `/home/turnerb/.PrusaSlicer/cache/Creality.idx-update.5933.download`
*   Trying 185.115.1.124...
* TCP_NODELAY set
* Connected to files.prusa3d.com (185.115.1.124) port 80 (#0)
> GET /wp-content/uploads/repository/PrusaSlicer-settings-master/live/Creality//index.idx HTTP/1.1
Host: files.prusa3d.com
User-Agent: PrusaSlicer/2.2.0-rc4+linux-x64
Accept: */*

< HTTP/1.1 301 Moved Permanently
< Server: nginx
< Date: Thu, 19 Mar 2020 16:30:48 GMT
< Content-Type: text/html; charset=iso-8859-1
< Content-Length: 398
[... 43 lines cut ...]
Host: octo.turnerb.com
User-Agent: PrusaSlicer/2.2.0-rc4+linux-x64
Accept: */*
X-Api-Key: B4740BA1A28E471EBE3BF2A92C0458EC

< HTTP/1.1 301 Moved Permanently
< Content-length: 0
< Location: https://octo.turnerb.com/api/version
< 
* Connection #0 to host octo.turnerb.com left intact
* Issue another request to this URL: 'https://octo.turnerb.com/api/version'
*   Trying 192.168.20.22...
* TCP_NODELAY set
* Connected to octo.turnerb.com (192.168.20.22) port 443 (#1)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
* SSL certificate problem: unable to get local issuer certificate
* Closing connection 1
[2020-03-19 12:30:50.977718] [0x00007f98c7da7540] [error]   OctoPrint: Error getting version: Peer certificate cannot be authenticated with given CA certificates:
SSL certificate problem: unable to get local issuer certificate
[Error 60], HTTP 0, body: ``

```

chemputer, 2020-03-19:
Okay, I decided on the nuclear option, to just bypass haproxy completely, and use :5000 (the port directly to octoprint) instead of using the SSL secured haproxy, so it's working, but not "as it should". It's just really, really weird that it works fine with haproxy on Windows, but not on Linux. Is that something that could be fixed?

bubnikv, 2020-03-19:
The developer who was responsible for the OctoPrint integration is no more
with us and I personally don't know the details of the Linux SSL
certificates. I don't know whether you are doing something wrong, or
whether PrusaSlicer is broken in general, or whether this is some distro
specific / binary incompatibility issue.

If you are a developer, it would be great if you can compile PrusaSlicer
from the sources and test it. It may be that the build will work correctly
as it it will match your distro. You may also try the non-appimage binary
build, the AppImage integration may be the culprit, see the following
discussion:
https://github.com/prusa3d/PrusaSlicer/issues/3388#issuecomment-601170118

Did PrusaSlicer OctoPrint over SLL worked at all in some older versions for
you?

Area5142, 2020-03-19:
A quick way to run the PrusaSlicer binary without AppImage container redirection is to extract the files from the AppImage:
`./PrusaSlicer-2.2.0-rc4+linux-x64-202003151142.AppImage --appimage-extract`

This will extract to a local directory. Go to:
```
cd squashfs-root/src/bin/bin
./prusa-slicer
```
This will run the binary outside the AppImage container.
I use this as test of 3D Mouse detection problem in #3388.

chemputer, 2020-03-19:
That's good to know, I wasn't aware that was functionality in appimage. That's pretty neat. 

I've tested both the prebuilt Linux binary on it's own, as well as the appimage. I haven't tried extracting it from the appimage, though. I can't imagine that would give different results, but I'll try it if it might help. 

As far as the previous question whether or not it worked in a previous version, I've only tested 2.1.1 and 2.2.0rc4, and neither of them seem to work with octoprint over SSL (at least via haproxy). The problem is the same on both. 

If you look at the log I posted from running the prebuilt Linux binary prusa-slicer from terminal with --loglevel 5, it shows that it's "unable to get local issuer certificate". 

Again, strangely, this works perfectly on Windows 10. I can't figure out which CA file it's using to do so. 

I have not tried building it from source yet, but I will try that to see if it makes any difference.

I'm happy to fire up a few VMs and test with different distros if that would be beneficial.

I'll also see if switching my haproxy.cfg back (temporarily, of course) to HTTP only and removing the SSL element 'fixes' anything (as it works fine when connecting directly to the OctoPrint instance over HTTP, which I'm kind of assuming at this point is probably just what I will have to live with), and given the logs from running it in the terminal I'm fairly sure it's an SSL issue. 

I appreciate the help from both of you.

bubnikv, 2020-03-19:
Note to myself.
I suppose we need to detect the distro and estimate location of the certificate authority file in runtime from our binary builds and we need to set

curl_easy_setopt(curl_handle, CURLOPT_CAINFO, "path_to_ca_file");

https://curl.haxx.se/libcurl/c/CURLOPT_CAINFO.html

If you compile PrusaSlicer with static dependencies, you may want to edit
deps/deps-linux.cmake
and replace 
        --with-ca-bundle=/etc/ssl/certs/ca-certificates.crt
with your location. I bet it will work then.

I believe the following is reasonably up to date:
https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md

Area5142, 2020-03-19:
@chemputer 
If you need to compile PrusaSlicer from source take a look at the shell script I posted on #3388 (BuildPS.sh.zip). It will install needed tools and build the binary on Linux Mint 19.3 x64 (Ubuntu).

bubnikv, 2020-03-20:
> I'm happy to fire up a few VMs and test with different distros if that would be beneficial.

You may certainly try that. Not many people connect to OctoPrint over SSL, and even less on Linux and yet even less on non-Debian based distros, so your is really an edge case.

I have asked @tamasmeszaros from our team to test the SSL connection to OctoPrint as he has a Raspberry PI at home and he is a avid Linux user. We may ask him to detect the CA file in runtime, though it feels like a possible security risk to me.

bubnikv, 2020-03-20:
What a hell

https://gitlab.com/probono/platformissues/blob/master/README.md#certificates

https://github.com/openssl/openssl/issues/7481

bubnikv, 2020-03-20:
You may likely fix it for yourself by symlinking your system's certificate file to the Debian way, for example on Fedora, you may try
ln -s /etc/pki/tls/certs/ca-bundle.crt /etc/ssl/certs/ca-certificates.crt
but it is quite an invasive hack.

chemputer, 2020-03-20:
Luckily, (and somewhat unfortunately, I think), I'm running a Debian-based distro (it's derived from Ubuntu 19.10, and Ubuntu is Debian based), and the ca-certificates.crt is in that exact location. 

You can see this in the very last bit of the log output. 

```
2020-03-19 12:30:50.838755] [0x00007f98c7da7540] [info]    OctoPrint: Get version at: http://octo.turnerb.com/api/version
*   Trying 192.168.20.22...
* TCP_NODELAY set
* Connected to octo.turnerb.com (192.168.20.22) port 80 (#0)
> GET /api/version HTTP/1.1
Host: octo.turnerb.com
User-Agent: PrusaSlicer/2.2.0-rc4+linux-x64
Accept: */*
X-Api-Key: B4740BA1A28E471EBE3BF2A92C0458EC

< HTTP/1.1 301 Moved Permanently
< Content-length: 0
< Location: https://octo.turnerb.com/api/version
< 
* Connection #0 to host octo.turnerb.com left intact
* Issue another request to this URL: 'https://octo.turnerb.com/api/version'
*   Trying 192.168.20.22...
* TCP_NODELAY set
* Connected to octo.turnerb.com (192.168.20.22) port 443 (#1)
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: none
* SSL certificate problem: unable to get local issuer certificate
* Closing connection 1
[2020-03-19 12:30:50.977718] [0x00007f98c7da7540] [error]   OctoPrint: Error getting version: Peer certificate cannot be authenticated with given CA certificates:
SSL certificate problem: unable to get local issuer certificate
[Error 60], HTTP 0, body: ``

```
Also, when I run `ls -la /etc/ssl/certs/ca-certificates.crt` on my machine, I get:
`-rw-r--r-- 1 root root 200061 Jan 10 11:41 /etc/ssl/certs/ca-certificates.crt`

Would it be extreme to suggest having an option to just accept whatever SSL cert you get? The issue seems to be when it's verifying it with the CA. That way, while it could be vulnerable to certain attacks (like MITM), it would still be encrypted. I would be cool with that. 

And if I can help test anything, let me know. Happy to help, but I have essentially no clue what I'm doing since I don't know the codebase.

vojtechkral, 2020-03-20:
@chemputer Try this on your command line:

    curl -v https://octo.turnerb.com/api/version

what output do you get?

chemputer, 2020-03-20:
```
 curl -v https://octo.turnerb.com/api/version
*   Trying 192.168.20.22:443...
* TCP_NODELAY set
* Connected to octo.turnerb.com (192.168.20.22) port 443 (#0)
* ALPN, offering h2
* ALPN, offering http/1.1
* successfully set certificate verify locations:
*   CAfile: none
  CApath: /etc/ssl/certs
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (OUT), TLS alert, unknown CA (560):
* SSL certificate problem: unable to get local issuer certificate
* Closing connection 0
curl: (60) SSL certificate problem: unable to get local issuer certificate
More details here: https://curl.haxx.se/docs/sslcerts.html

curl failed to verify the legitimacy of the server and therefore could not
establish a secure connection to it. To learn more about this situation and
how to fix it, please visit the web page mentioned above.

```

And yet, if I connect via Chromium, I get: 
`{"api":"0.1","server":"1.4.0","text":"OctoPrint 1.4.0"}`
Which I assume is because web browsers usually have their own CA's built in.

vojtechkral, 2020-03-20:
> Which I assume is because web browsers usually have their own CA's built in.

If I remember they store additional certs somewhere in `$HOME`.

In any case, this seems like an issue with the system-wide cert store. Maybe it's not up to date...

chemputer, 2020-03-21:
Good news everyone, I figured it out! It's **NOT** an issue with PS! (I think you guys probably knew that.) It turns out that it's a common issue with Let's Encrypt certs that they aren't accepted by curl/wget, etc., because the intermediate CA is missing, and not provided by the server. The root is of course in the system's CA, but the intermediate is not, and the website is supposed to provide it. The reason the browsers work is because other sites, who aren't run by incompetent amateurs like myself, include those intermediate CA's, and they're cached, so it verifies it that way.

[This website](https://www.meshcloud.io/2017/04/18/pem-file-layout-for-haproxy/) was helpful, for anyone who has the issue in the future. While it says there's a strict order, I found it's not that strict. Using LE certs from certbot, you have to concatenate the files as such:
[domain].crt [domain].key [domain].ca > ssl_cert.pem, and use ssl_cert.pem in your haproxy/nginx/apache config. They may be different for the different servers, but this works for me. 

I also noticed that this was actually already provided in the form of [domain].all.pem, but I'm unsure if that's standard certbot as I'm using my pfSense router to do all the certificate stuff, then just copying as needed to where I need them. 

Anyway, I appreciate all of your help, because I'm an idiot, and it took me looking into it at several different angles to find the fix, and your questions and suggestions nudged me in the correct direction.

That said, I did discover that if you use the option curl -k (or --insecure), then it will just accept whatever it SSL cert it receives. Given that this is just for 3D printing, a little checkbox next to the OctoPrint area that allows curl to accept "invalid" certs would be awesome. It would still be TLS, but just not checked against a CA. There's some documentation on it a few lines down in [curl's online documentation](https://curl.haxx.se/docs/sslcerts.html). I'm not sure how hard that would be to implement, but if it's simple (a checkbox to add an argument?), I'm sure the few edge-cases like me would appreciate it. Although going down the rabbit-hole of "doing it right" is always an option.

Also, @Area5142 I noticed a small issue with your script (absolutely awesome, by the way, thank you!), and made two tiny tweaks that will let it work on pretty much any debian-based system. 

I changed it to install software-properties-common, which installs the add-apt-repository/apt-add-repository command on systems that don't have it already, and then changed the apt add-repository to apt-add-repository, which works on all debian-bases systems. I think apt add-repository is Linux Mint specific? I'm not sure, as I couldn't find any documentation on it. [This askubuntu.com question](https://askubuntu.com/questions/493460/how-to-install-add-apt-repository-using-the-terminal) was helpful.
```
    
    # Install package that adds apt-add-repository
    # See: https://askubuntu.com/questions/493460/how-to-install-add-apt-repository-using-the-terminal     
    echo "Installing apt-add-repository..."
    sudo apt update
    sudo apt install software-properties-common -y

    # Install git:
    echo "Installing git..."
    sudo apt-add-repository -y ppa:git-core/ppa
```
Here's the modified script. 
[BuildPS-modified.sh.zip](https://github.com/prusa3d/PrusaSlicer/files/4362502/BuildPS-modified.sh.zip)
It was really helpful in getting PrusaSlicer to build (although, in the end it didn't make any difference), so it would be awesome if it could be added to the [How to Build - Linux Guide](https://github.com/prusa3d/PrusaSlicer/blob/master/doc/How%20to%20build%20-%20Linux%20et%20al.md), to help those with Debian-based Distros, even if it might not work 100%, it should hopefully streamline the process.

I'm not sure if I should close this or not, so I'm leaving it open. It would be nice to have that curl -k/ curl --insecure checkbox option, but that's just me, and I'm sure there are higher priority things out there. Feel free to close it if you think appropriate.

Thanks again!

bubnikv, 2020-03-21:
Great Ben that you have found out.

I believe everybody could edit https://github.com/prusa3d/PrusaSlicer/wiki
It would be great if you could create a new page there describing your
findings on how to configure OctoPrint / haproxy with SSL.

Area5142, 2020-03-21:
@chemputer 
Thank you for the additions to the BuildPS.sh script.

I found some other small errors and have now updated the script:

1) Changed build directory from _build_ to _Build_ because _Build_ is already excluded in .gitignore.
2) Added install of _software-properties-common_ to add-apt-repository/apt-add-repository command (thanks to @chemputer).
3) Added more documentation in the script (comments).

The script is far from a complete solution to compile PrusaSlicer from source on Linux, but a starting point to get _a first compile_. Use at your own risk... ;-)

[BuildPS version 0.1.zip](https://github.com/prusa3d/PrusaSlicer/files/4363426/BuildPS.version.0.1.zip)

@bubnikv 
I am new to writing wiki pages, but can make a page with the script to help others compile under Linux Mint (Ubuntu based distributions), if it is OK with you.

bubnikv, 2020-03-21:
> I am new to writing wiki pages, but can make a page with the script to
help others compile under Linux Mint (Ubuntu based distributions), if it is
OK with you.

Any help to the community will be greatly appreciated.

bubnikv, 2020-03-21:
The upstream Slic3r has a nice page on how to compile on various distros
using the distro packages.
https://github.com/slic3r/Slic3r/wiki/Running-Slic3r-from-git-on-GNU-Linux

Frankly I am not sure how useful that would be for PrusaSlicer, as we do
not test using distro packages, but we rather rely on carefully tested
variants of libraries compiled statically.

Abrynos, 2023-12-10:
I've had a similar issue with mainsail. The solution was to disable TLSV1.3 in my nginx configuration.

```diff
- ssl_protocols TLSv1.3;
```

@chemputer I'm not sure on the options you have with octoprint, but could you try that?

chemputer, 2023-12-11:
> I've had a similar issue with mainsail. The solution was to disable TLSV1.3 in my nginx configuration.
> 
> ```diff
> - ssl_protocols TLSv1.3;
> ```
> 
> @chemputer I'm not sure on the options you have with octoprint, but could you try that?

Yeah, keeping in mind that this was 4 years ago and I no longer have the same setup, but I did something similar. It's not a solution, though, nor should anyone pretend it's anything other than a temporary workaround.

Abrynos, 2023-12-12:
> this was 4 years ago

Da***t. I am not used to tickets staying open that long so it seems a bump was needed anyways 😉 

> It's not a solution, though, nor should anyone pretend it's anything other than a temporary workaround.

I fully agree with you, but honestly I have no idea where I would go looking in the code to fix it and open a merge request.

Abrynos, 2023-12-12:
Oh btw you can remove the `on Linux` part from your original ticket title. The same problem is present on Windows. I've tested both.

chemputer, 2023-12-13:
> Oh btw you can remove the `on Linux` part from your original ticket title. The same problem is present on Windows. I've tested both.

Not that I don't trust you, hell, I even believe it's not working for you on Windows, but I have two computers where I processed files one was, at the time, Windows 10, now 11, the other was Pop! OS, now Debian, but I do distinctly remember getting rather frustrated that it was working fine for me on the Windows machine, same software and all, connected in the same way to the same OctoPrint instance on I _think_ a Rock64 board, and the Windows worked just fine, but without either modifying the Haproxy setup or else installing nginx and just using _that_ in front of the HAProxy (it was used for convenience for the webcams more than anything else, I was following a guide at the time, too.)

Whether or not it works now, with a different setup, I don't know. I'm not sure if removing the tag when I know that with that version and setup and everything it didn't work, etc. would actually be useful or not. Frankly I'm not sure what the typical protocol is here, like you, I'm not used to tickets staying open this long, but also, just rereading over this ticket after so long, I realized that I found out that it didn't seem to be an issue with PrusaSlicer, but rather some particular issues with Let's Encrypt's (i.e. the backbone for CertBot) Intermediate CA not being present and having to recreate the PEM file for HAProxy. I _think_ that this ticket should be closed since that seemed to be the "fix" and pointed in a direction that was not related to PS, but I'm honestly not sure. So I suspect that whatever is the cause of your issue, if it's effecting Windows too, is either similar but I don't know enough about your issue to be able to tell you, or it's something different entirely. I'd suggest making a new issue, mentioning this issue can't hurt, but giving all the details a new issue would want is only going to be helpful.

[This is the comment](https://github.com/prusa3d/PrusaSlicer/issues/3857#issuecomment-602001582) I made that I just re-read that got me back up to speed about what ended up causing the issue and the fix. See if that works for you, if not, then definitely open a new issue! 

I'm not sure whether to close this one or not, as best I can tell it's not a PrusaSlicer issue, as weird of an issue as it may present as in my case.

closed 2026-09-29 by prusaslicer-bot[bot]