prusa3d/PrusaSlicer #3859: Bridge max linear length must be capped in "don't support bridges"

nordurljosahvida, 2020-03-18:
### Version

2.2.0-rc4, all previous versions that i know of have had the same issue

_Use `About->About PrusaSlicer` for release versions_

### Operating system type + version

Ubuntu 18

### 3D printer brand / version + firmware version (if known)

MK3S MMU2S

### Behavior

* _Describe the problem_

The "don't support bridges" option fails to take into consideration the length of a bridge.
Countless times our prints require us to uncheck this advanced option to simply support anything in even very simple objects. I don't think we can expect non-technical users to know to go into advanced and uncheck this, in addition to the fact that in the same print there might be other - actual - bridges that would get supported for no reason.

I've executed a print one of the teachers in our client school had sliced before I advised her to uncheck the box:

[image]
[image]

* _Steps needed to reproduce the problem_

Print anything with bridges longer than a few cm

* _Expected Results_

PS actually defining a bridge when it is reasonably printable without supports, that meaning only up until a certain lenght. A bridge that crosses the entire buildplate is not a bridge in this sense, and must be supported.

* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

No, a related FR is here but I've opened this as an issue to gather attention to the problem.

https://github.com/prusa3d/PrusaSlicer/issues/1970

#### Project File (.3MF) where problem occurs

[bridge_issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4348682/bridge_issue.zip)

Wallby, 2021-10-03:
Shouldn't this be titled "Feature request..?". I.e. the requested feature is to set the min bridge length to generate supports for instead of being either no supports for bridges, or supporting all bridges.

Wallby, 2021-10-03:
I think this scenario is an example of when a switch is undesirable..
[image]
[image]

nordurljosahvida, 2021-10-04:
@Wallby it is firstly an issue: the don't support bridges cannot be simply controlled by a user not in expert mode, and it cannot be possible that we can't print an empty cube without the user having to know about bridging, max lenght, cooling, machine capabilities, etc. If anything, we should err on the side of supporting excessively.

FR: add the possibility to set the custom lenght, but in the mean time let's make a default value!

closed 2026-09-29 by prusaslicer-bot[bot]