prusa3d/PrusaSlicer #3864: 3Dconnexion

stedoc, 2020-03-18:
### Version
_Version of PrusaSlicer used goes here_

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

FidelCapo, 2020-03-19:
Can you please fill in form? We need to know which OS and version of PrusaSlicer you are using. Thank you.

jweaver100, 2020-03-19:
I was intrigued by this as I havn't seen this behavour myself... I just tried and do not get this issue.. I clicked all around the UI and moved the 3D mouse and was unable to change any settings.

I am using RC3 with the modified camera fix (RC3+5-et-free-camera-win64) if that helps

bubnikv, 2020-03-19:
Would you please test with rc4?

The 3dconnexion driver sends scroll wheel events to the application. I wonder whether you can disable that in the 3dconnexion driver. We filter out these scroll wheel events in the 3d scene, but if any other UI component that accepts mouse scroll events is focused and you touch the 3dconnexion ball, you may get the scroll events I suppose, which will change combo box selection and similar.

@FidelCapo could you test that please?

bubnikv, 2020-03-19:
You may likely convince your driver to stop sending these scroll wheel events if you convince the driver that PrusaSlicer reads the device directly (raw input), but I did not try it.

https://www.3dconnexion.com/forum/viewtopic.php?t=7541#p32503
https://www.3dconnexion.com/forum/viewtopic.php?t=9774#p37824

FidelCapo, 2020-03-19:
I can reproduce it on Win in PrusaSlicer RC4. Workaround can be turn off all items in the 3Dconnexion settings as shown on the screeshot.

[image]

jweaver100, 2020-03-19:
I just tried RC4 and I can't reproduce it.. That said, I am using an old 3D mouse (SpacePilot) and old drivers (as the SpacePilot is not supported in the recent drivers).

neophyl, 2020-03-19:
I'm running rc4 on windows 10 with a new spacemouse compact bought at Christmas and I cant reproduce the issue either.  I have tried all axis in the plater, while i have the numeric fields selected, in the drop downs, and in all 3 settings tabs (print, filament and printer) and at no point was it changing anything. 

I know you don't use the 3d connection settings but the spacemouse advanced settings as pictured by FidelCapo has nearly all of them ticked on mine and isnt causing me any problems.

bubnikv, 2020-03-20:
For the record, I know @enricoturri1966 has invested considerable amount of time to suppress these spurious scroll events at least in the 3D scene. Whether you get the scroll events delivered to the combo boxes depends on the version of the 3DConnexion driver I suppose or maybe on the alignments of the planets.

stedoc, 2020-03-20:
Thanks everyone, but I think I solved it I reinstalled the drives and set the 3d mouse as suggested seems to work
thank you all for your help

bubnikv, 2020-03-20:
I just verified on my setup. The 3DConnexion sends scroll events to a control, over which the mouse cursor hovers, independently from whether the control has focus (one clicked with the mouse on it before) or not. This is very unexpected.

So the reasonable option is really to disable the scrolling emulation for PrusaSlicer in the driver.

bubnikv, 2020-10-16:
The scroll events seem to be suppressed once the PrusaSlicer application is recognized by the 3DConnexion driver on Windows.
That happens once a proper prusa-slicer.xml file is copied into the 3DConnexion system wide configuration directory. The latest 3DConnexion driver already does that and we have verified that these spurious scroll events are gone.

Closing.

closed 2020-10-16 by bubnikv