prusa3d/PrusaSlicer #3865: Drilling hole failure causes all previously valid holes to fail in SLA Hollow and Drill Gizmo

kenshinvt, 2020-03-19:
### Version
_Version of PrusaSlicer used goes here_
2.2.0-rc4

### Operating system type + version
_What OS are you using, and state any version #s_
Win10 64bit latest updates

### Behavior
* _Describe the problem_
If a hole is added that is invalid/errors in the hollow & drill gizmo, then sometimes the entire model will no longer drill properly until all holes are removed--even if they were valid previously.

* _Steps needed to reproduce the problem_
An STL file is loaded.  Using the hollow/drill gizmo, the object is hollowed and one hole (#1) is placed.  The preview hollow/drilled works and shows the hole.
[image]

Another hole (#2) is added.  This hole is, for whatever reason, not valid and the failed drill error message appears.
[image]

To remedy this, the new "bad" hole #2 is removed and the preview done again.  It is expected that--since the only hole remaining is the first one that was fine--the render will now be ok.  However, it now fails even on this previously "good" hole #1.  Also the model no longer appears hollow and is solid.
[image]

* _Expected Results_
Removing the "bad" hole will return to the state before it was added.

* _Actual Results_
Previously added holes are all treated as failed/errors.

_Is this a new feature request?_
No

_Notes_
- This has happened across multiple different, unrelated STL files
- This does NOT always happen.  Sometimes the model will render as expected (i.e. as it was in the prior step).  I'm not sure what the changed variable is that causes it to happen in certain instances.  However, it is happening frequently enough in projects that I felt an issue submission was warranted.
- Lastly, while the the hollow and drill feature is fantastic to finally have in the slicer, I sincerely hope there is some way to make the drill feature more reliable in general.  I see many failures on drilling on good, error-free STL models and it is not clear in advance which drill attempts will succeed or fail.  If there is any way to make the drill feature less sensitive or provide better feedback (before a lengthy render) that a failure would occur.. it would be really excellent!  Thank you!

tamasmeszaros, 2020-03-19:
Hello,
could you provide an stl file (a 3mf would be even better) on which the fails occur? Thank you in advance.

kenshinvt, 2020-03-19:
Hello, this has happened to me on a variety of files.  Most of them have been complex/larger files, but none of which have had errors or been in need of repair.  Hopefully it is possible to reproduce it when you see a drilling error.  I do not have the rights to distribute the above file shown in the example pictures.  However, if that file is specifically needed to recreate then please let me know and I will ask the author if I can share it privately.  Thanks

closed 2026-09-29 by prusaslicer-bot[bot]