prusa3d/PrusaSlicer #3866: Feature Request - Ability to insert exclusion zones on the print bed to avoid damage to PEI

ChrisVahi, 2020-03-19:
### Version 
PrusaSlicer 2.1.1
Windows 10 Home version 1903
Prusa i3 mk3s mmu2

### Behavior
* _Describe the problem_
Due to printing in the same place on the bed wear spots have formed with damage to the PEI sticker. I am able to manually move parts out of the way but this is tedious, especially for prints with many parts. I request a method to define the damaged locations in the Bed Shape dialogue and have the arrange routine avoid those locations automatically. 

*_Is this a new feature request?_ 
Yes, but it is similar to the desired result of issue #320 

#### Project File (.3MF) where problem occurs
All files uploaded are affected by this condition.

Area5142, 2020-03-21:
It can be done in PS this way:

1) Make a part matching the _"non printing zone"_ of the heat bed.
2) Place it in the right spot on the heat bed.
3) Select all parts (Ctrl+A) and deselect _"non printing zone"_ part (Ctrl+Click).
4) Select arrange selected (Shift+A). This will arrange all other parts around the zone.
5) Disable _"non printing zone"_ part before generating g-code.

It can be done but is a little complicated.
1+2 could be made once and saved in 3mf file for each print sheet.

Would be nice to have a lock function to keep manual placed objects from being moved, even if they are disabled. It would make it easier to arrange parts for printing and accidentally being moved.

ChrisVahi, 2020-03-21:
> It can be done in PS this way:
>
> 1. Make a part matching the _"non printing zone"_ of the heat bed.
> 2. Place it in the right spot on the heat bed.
> 3. Select all parts (Ctrl+A) and deselect _"non [...]

Nice workaround, though yes, it is tedious. I'll try it for now but Prusa team, can we work on a way to automate this?

Area5142, 2020-03-22:
To only arrange selected parts use **Shift+A**, not just A or the tool button.
[image]

foreachthing, 2020-03-22:
I wanted to be smart .. but, nope!
[image]
Arrange does nothing.

Area5142, 2020-03-22:
I see no problem:
[image]

Select the parts you want to be arranged (on your picture only one is selected) before pressing Shift+A (all parts must be enabled).
After that, disable the part not to be printed (bad spot).

foreachthing, 2020-03-22:
@Area5142, well, I tried to load a modified stl, which can easally be changed in blender..

Area5142, 2020-03-22:
I'm not sure, what the problem is. If all parts are STL and imported as separate models it works.

foreachthing, 2020-03-22:
Tried to change the bed-stl. Not a part, but the printer plate. No need to select all and unselect this other geometry. Just load your parts, arrange and done. Unfortunately, this does not work correctly.

SolidRick, 2020-03-22:
I edited my svg bed graphics (...\PrusaSlicer\resources\icons\bed\mk2.svg) with Inkscape, adding some red spots, see below:
[image]
In this way I can manually place the models out of the damaged spots

FlailAway, 2020-06-24:
> It can be done in PS this way:
> 
>     1. Make a part matching the _"non printing zone"_ of the heat bed.

I just posted about this issue [(map bad spots)](https://github.com/prusa3d/PrusaSlicer/issues/4431#issuecomment-648701046) ) and was pointed back here.

But this approach is per item. The lump in my bed is always there so I move it away except on those few times I forget. I suffer the frailties of being human. Me moving the part first thing is easier than the process suggested here if this process cannot be automated, what's the point. <--rhetorical :)

BMacours, 2021-05-13:
I second this feature request, with the ability to switch from a bed to another from within PS as each bed has its own "dead" zones. Thanks a lot!

Metti93, 2022-08-19:
/push

a Tool-Button for defects in the Bed, that would be really nice.

Tupson444, 2023-05-06:
This would be useful for printers that use bed clips. I have made custom bed textures with clips indicated, so I can manually avoid placing anything there (after slicing I also check if skirt is passing through it). But it would be better if we could add exclusion zones, so that arrange always avoids those areas.

Also (maybe), the exclusion zones could have a specified Z as well. When the print head is under the Z value, all travel moves should avoid exclusion zones, while it can safely travel over it when at higher Z.

I could open a new feature request for avoiding bed clips, but I'm just commenting here since it's technically the same as this request (users would define the size of the exclusion zones, for whatever reason they want to avoid movement through them).

Tupson444, 2023-08-24:
This might be solved by new custom bed shape introduced with version 2.6.1-rc1. However, it still has some limitations, see #11171.

KyleMaas, 2025-04-29:
I'd love to see this as well.  I have been manually arranging parts on my damaged borosilicate glass bed to try to avoid some damaged areas, trying to hold off on replacing the glass for as long as possible, and I figured I'd do a search to see if this was possible.  This would be a really handy labor-saving and time-saving feature by prolonging the duration between bed replacements.

Tupson444, 2026-09-13:
still wanted