prusa3d/PrusaSlicer #3870: Brim doesn't consider Elefant Foot Compensation

CNCKitchen, 2020-03-20:
### Version
2.2.0 rc 5

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
When using a brim + elephant foot compensation the brim won't touch the part anymore making it basically useless.
==> Elephant foot compensation should also move the brim closer to the part.

This has already been discussed here (https://github.com/prusa3d/PrusaSlicer/issues/2156) but still seems to be happening.

[image]

bubnikv, 2020-03-20:
This is on purpose. The elephant foot compensation compensates for the
squish, which is not visualised in the G-code preview. In reality, the gap
will be filled in by the squished out material. It is true, that the brim
now has a weaker connection to the object than before, but most likely this
is welcomed than a problem.

See the following discussion as well.
https://github.com/prusa3d/PrusaSlicer/issues/3779

gihad, 2020-04-20:
I just encountered this and came to the conclusion that the elephant foot compensation does not compensate enough. The brim was still not connected to the object (my first layer z height is properly set). To fix this one has to disable elephant foot compensation (set it to 0) and than brim works again.

fiveangle, 2020-04-20:
Since EFC is user-defined while the actual physical squish of the filament is physics-defined, only if the user-defined EFC matches the actual physics-defined squish, will the brim provide adequate connection to the first layer.

I don't think assuming EFC matches physical squish is a viable solution without also adding a training component: "If brim fails to connect to model, your EFC is to large and should be reduced."

This seems more trouble than simply accepting that the brim may be "too connected" if it is subjected to the same EFC since if the user enables the Brim feature, they are already accepting that they must remove the brim from the model (and all the downsides that it entails).

mikolaszuza, 2020-08-03:
This might be a good candidate for a notification with action (which are coming in 2.3) when you turn on Brim.
"Brim will have a weaker connection to the model with Elephant foot compensation turned on, **Turn Elephant foot compensation off?**"

neophyl, 2020-08-03:
I prefer Supermerills recent implementation from SuperSlicer.  With that the brim is moved in with the objects layers to take into account the elephants foot compensation so is always connected to the part but the brim settings have a brim offset value so you can explicitly set the object-brim distance to make it as weak or strongly connected as you like.

That way they are in effect independent of each other.

Andy-Prusa, 2020-10-03:
If I can just point out that I have had four failed tall prints before I self diagnosed this problem. Etc was set at 0.2, and looking at the brim immediately after its printed looks fine. But near the top of the print when the moment of force is greater it always seperated. I tried different printing temps to no avail, and it was only on close inspection of the sliced model that I saw the gap. Changed etc to 0.1 and it printed perfectly. A warning about this would certainly save a lot of heartache. 

P.s. I had tuned my printer to the best of my ability before this  and measurements showed an Efc of 0.2 to be adequately accurate but too much for good brim adhesion

marcantoinebeaulieu, 2021-01-13:
Still happens in 2.3.0. I wish we could get at least a popup reminder that asks to set it to 0.

fiveangle, 2021-01-24:
Can someone at Prusa3D please mark this as a blocker for the next release ?

The only way a brim will work currently is if we disable elephants foot compensation. If we forget,which is like almost every time because E.F.C. is the default, then we get to waste 10min scraping the print, re-cleaning the bed, go back and reslice, sneakernet back to the printer, re-wait for heating, and then print gets restarted. It's kinda silly and no amount of arguing semantics is going to change this reality *sigh* [image](https://user-images.githubusercontent.com/2219447/105643475-ee13e380-5e44-11eb-852d-37b8af7fe6c5.jpeg).

Andy-Prusa, 2021-01-25:
> Can someone at Prusa3D please mark this as a blocker for the next release ?
>
> The only way a brim will work currently is if we disable elephants foot compensation. If we forget,which is like almost [...]

Here here

lukasmatena, 2021-08-12:
In PrusaSlicer 2.4.0-alpha1, the brim will always touch the object, regardless of EFC setting. Additional 'brim_offset' parameter will allow to fine tune the gap to anyone's liking.

Closing.

closed 2021-08-12 by lukasmatena

Uko, 2021-08-12:
Thank you very much!