prusa3d/PrusaSlicer #3871: Option: First Layer Speed also applied to second layer

jigsawgirl, 2020-03-20:
Being able to set the "First Layer Speed" setting to also apply to the second layer.

I have set the First Layer Speed to 20mm/s to ensure a well attached first layer. This works very well and I have no issues with bed adhesion at normal speeds. However I often run prints with speeds of 78mm/s for the rest of the print. The faster speed is reliable once it gets to the 3rd layer but will often pull off part of the first layer. This only happens intermittently and on parts of the model that have a small contact area on the build plate.

Being able to reduce the speed on the second layer as well as the first would help reduce issues when running at high speeds.
  
### Version
PrusaSlicer 2.1.1

### Operating system type + version
Windows 10 Pro 64bit

### 3D printer brand / version + firmware version (if known)
Ender 3 Pro

### Behavior
Feature Request

_Is this a new feature request?_
Yes

rtyr, 2021-02-08:
It is possible by using Height range modifier. Is it not easy enough?

bubnikv, 2021-06-03:
No response, closing.

closed 2021-06-03 by bubnikv

valterm, 2025-02-15:
I'd like to echo the request.

My printer consistently messes up the second layer if printed at higher speeds, but is perfectly fine after layer 3.

Using a height range modifier is a functioning workaround, but 
- it needs to be done each time
- it needs to be done for each model on the bed

Adding something like "Apply first layer settings to X layers" would be a good QoL improvement.