prusa3d/PrusaSlicer #3872: 2.2.0 RC5 - Height range modifiers not working

Itox001, 2020-03-20:
### Version
2.2.0 RC5

### Operating system type + version
Windows 10

### Behavior
Height range modifiers are not being applied. When setting the external perimeter speed for a single wall cube, the speeds remain the same throughout the print. This can be seen on the preview, and I also reviewed the Gcode and can see that this is the case:
[image]

The legend has the speeds correctly from 20 to 50 mm/s, but the speed remains in all the print to 20mm/s.

#### Project File (.3MF) where problem occurs

[20mmTestCube_repaired.zip](https://github.com/prusa3d/PrusaSlicer/files/4361373/20mmTestCube_repaired.zip)

Itox001, 2020-03-20:
Doing the same using modifiers works as expected:
[image]

rtyr, 2020-03-20:
It is related to "clip multipart objects" parameter. Enable this parameter to fix the issue.

It will be enabled by default in PrusaSlicer 2.2.0 final. See https://github.com/prusa3d/PrusaSlicer/issues/2803. Closing.

closed 2020-03-20 by rtyr