prusa3d/PrusaSlicer #3873: Little bug

jurstyx, 2020-03-20:
### Version
2.1.1+win64

### Operating system type + version
Windows 10 Pro 1903

Hello, folks,

first of all I would like to tell you that I appreciate your work very much and I am very enthusiastic about the prusa brand.

I found a little bug. After slicing an object, if you drag and drop another object into the prusaslicer, you suddenly don't see an object anymore. If you then sliced it again, it will appear again. It's not very bad, but I thought I better tell you.

Thanks,
Daniel

YuSanka, 2020-03-20:
Thanks for your report. It seems you are trying to add (by dropping a file) new objects to the scene while the 3D scene is in "Preview" mode. You can check the current mode in the left bottom of the main window:
[image]
In this mode you see objects that are already sliced. When a new unsliced object is added the whole sliced scene is invalidated. As the result, you see an empty scene. Please, switch back to the "3D editor view" mode before or after the drag and drop:
[image]
Then you can switch to the "Preview" mode again and the scene will be re-sliced.

We understand that the current behavior is a little confusing. 
* We can make it so that the scene is automatically resliced when a new object is drag and dropped while in "Preview" mode.
* Alternatively, we can make it so that Slicer is switched automatically to the "3D editor view" mode on drag and drops.

We need to decide which option is the best in this case.

Evovita, 2023-06-02:
In `PrusaSlicer-2.5.2` I can reproduce this bug, but not in `PrusaSlicer-2.6.0-beta3`. Instead, the mode changes to 3D Editor view mode on drag and drop - so it's possible a fix was implemented as described by @YuSanka and this has now become a stale issue?

rtyr, 2023-09-12:
No response. I am closing the issue.

closed 2023-09-12 by rtyr