prusa3d/PrusaSlicer #3874: 2.2.0-rc4 Incomplete Slicing

BirkBinnard, 2020-03-20:
### Version
2.2.0-rc5

### Operating system type + version
Windows 10/64 current build

### Behavior
Incomplete slicing. 

2 screenshots show before and after images.

STL file loads as Manifold. 3D Builder reports no errors in file. Craftware slices file OK.

[image]

[image]

[postlight1.zip](https://github.com/prusa3d/PrusaSlicer/files/4361888/postlight1.zip)

lukasmatena, 2020-03-20:
Could you please enclose a 3MF? I tried to slice your model with standard MK3S profile and all seems to be fine.

BirkBinnard, 2020-03-20:
Very interesting - but I figured this was an obscure one. Attached is 3MF.
[postlight-3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4362055/postlight-3mf.zip)

lukasmatena, 2020-03-20:
The 3MF that you posted was not generated by PrusaSlicer. What we need is that you open the model in PrusaSlicer, set everything so that you see the issue and save the project as 3MF (File->Save Project As). This will save not just the model, but also complete config that you used for slicing it. It is then more likely that we willl be able to reproduce your problem.

BirkBinnard, 2020-03-20:
Uh oh!
While all this was going on I discovered I made a design error in the part. So I fixed that and re-saved it as a 3MF file.  That sliced OK after 3D Builder fixed 4 errors. I also read the Prusa writeup on 3MF and it's pretty clear that is a much better file type to use. I did verify that Thingiverse supports the 3MF format.

I re-saved my updated geometry as STL and sliced that with no problems also. So my bottom line on this is there must have been some sort of obscure geometry problem with my initial  design. 

My bad - sorry about that. I'll close thing issue and chalk it up to the age-old User Error category.

Thanks for your timely response and getting me educated about 3MF.

BirkBinnard, 2020-03-20:
User error - please disregard.

closed 2020-03-20 by BirkBinnard