prusa3d/PrusaSlicer #3876: 2.2.0-RC5 Does not slice

Grogyan, 2020-03-21:
### Version
2.2.0-RC5

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

### 3D printer brand / version + firmware version (if knownn
Generic RepRap I3 MK2.5 Running Marlin 2

### Behavior
* _Describe the problem_
Slicing a model results in no sliced gcode.

* _Steps needed to reproduce the problem_
Using a fresh clean copy of PrusaSlicer 2.2.0-RC5
Ensure Virus Scanner is disabled, or allow PrusSlicer access to write data to computer
Update Printer profile, including start and end gcode
Load in a 3D Benchy, which has been mesh fixed using Windows Builder, saved in .3mf
Slice model
No gcode generated at all

* _Expected Results_
Load in a model, 3D Benchy in this case.
Slice
Gcode generated for printing
* _Actual Results_
No Gcode is generated

[PrusaSlicer_config_bundle.zip](https://github.com/prusa3d/PrusaSlicer/files/4362301/PrusaSlicer_config_bundle.zip)

Grogyan, 2020-03-21:
No Errors were provided.

Restarted PS 2.2.0-RC5 multiple times
Still no errors.

Error presented after restarting over 10x of PS2.2.0-RC5 of initial layer higher cannot exceed nozzle diameter
Which showed that PS2.2.0-RC5 did not automatically adjust or notify me of this parameter when the settings were saved, as the nozzle diameter was changed from 0.4mm to 0.3mm

bubnikv, 2020-03-21:
So you expect the install wizard to adjust the numbers or give you a
warning?

If you follow the error message, is your problem solved?

Grogyan, 2020-03-21:
I would expect an error to be produced.

As stated, no error came up
Not even restarting PrusaSlicer 2.2.0-RC5 several times.

About an hour later I did get an error popup. 
I would say that the problem isn't solved, if no error pop up comes up for an hour.

But it should be a warning, not an error, for "Initial layer must be less than nozzle diameter"

Now looking at the first Benchy print with the same settings as I use on Cura, and it is awful.
There is certainly some weird stuff occurring during print.  But that is a separate problem, that haven't yet located the source of the problem.

rtyr, 2020-05-18:
I tried your settings in 2.2.0 final. Error message is produced correctly, so I believe it si fixed. Closing.

closed 2020-05-18 by rtyr