prusa3d/PrusaSlicer #3877: GLCanvas3D.cpp compile error on Linux

Area5142, 2020-03-21:
### Version
Git tag: version_2.2.0-rc5

### Operating system type + version
Linux Mint 19.3 x64 (Ubuntu)

### Behavior
Compile error in GLCanvas3D.cpp:
```
src/PrusaSlicer/src/slic3r/GUI/GLCanvas3D.cpp:3343:5: error: ‘m_tooltip’ was not declared in this scope
     m_tooltip.set_in_imgui(false);
     ^~~~~~~~~
```
### Suggested solution
Surround the two _m_tooltip.set_in_imgui()_ calls with conditional compile, as done in other places in the source:
```
#if ENABLE_CANVAS_TOOLTIP_USING_IMGUI
    m_tooltip.set_in_imgui(false);
#endif // ENABLE_CANVAS_TOOLTIP_USING_IMGUI
    if (imgui->update_mouse_data(evt)) {
        m_mouse.position = evt.Leaving() ? Vec2d(-1.0, -1.0) : pos.cast<double>();
#if ENABLE_CANVAS_TOOLTIP_USING_IMGUI
        m_tooltip.set_in_imgui(true);
#endif // ENABLE_CANVAS_TOOLTIP_USING_IMGUI
```
With this change the full source compiles without error.

bubnikv, 2020-03-21:
I think there may be an issue with your current source code or compilation
state.
Try pulling a fresh code and recompiling from scratch.

Area5142, 2020-03-21:
Somehow the define _ENABLE_CANVAS_TOOLTIP_USING_IMGUI_ in _Technologies.hpp_ evaluated to **false** and triggered a compilation error (should not happen).

You are right, a fresh clone of the source made the compiling work again as _ENABLE_CANVAS_TOOLTIP_USING_IMGUI_ now evaluates to **true** and _m-toolset_ gets defined ([here](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GLCanvas3D.hpp#L498)).

I still think the lines [here](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GLCanvas3D.cpp#L3343) and [here](https://github.com/prusa3d/PrusaSlicer/blob/master/src/slic3r/GUI/GLCanvas3D.cpp#L3346) should be surrounded with conditional compiles as done  elsewhere in the source. If not, compile will still fail when _ENABLE_CANVAS_TOOLTIP_USING_IMGUI_ evaluates to **false**.

I know this is a minor error ;-)

bubnikv, 2020-03-21:
All the non-ENABLE_CANVAS_TOOLTIP_USING_IMGUI  will be removed by @enricoturri1966 while he gets confident about the new code paths.

Area5142, 2020-03-21:
That will solve the issue and make the code more readable.

lukasmatena, 2020-05-18:
The `ENABLE_CANVAS_TOOLTIP_USING_IMGUI` #define was removed in 9487676680d7243b8520408e0989b7b0bc3e6bb8. Closing.

closed 2020-05-18 by lukasmatena