prusa3d/PrusaSlicer #3878: Mini Firmware uses Junction Deviation / PrusaSlicer uses Jerk

dahuafschmied, 2020-03-21:
### Version
2.2.0 RC5

### Operating system type + version
independend

### 3D printer brand / version + firmware version (if known)
Prusa Mini 4.0.2-4.0.5RC

### Behavior
Prusa Mini Firmware (4.0.2 - 4.0.4) uses Junction Deviation
shown with M503

```
Recv: echo:Advanced: B<min_segment_time_us> S<min_feedrate> T<min_travel_feedrate> J<junc_dev>
Recv: echo:  M205 B20000.00 S0.00 T0.00 J0.02

```
But PrusaSlicer uses Jerk settings:

`M205 X8.00 Y8.00 Z2.00 E10.00 ; sets the jerk limits, mm/sec
`
the values from slicer are ignored by the firmware.

bubnikv, 2020-03-21:
Duplicate of https://github.com/prusa3d/PrusaSlicer/issues/1089

see especially this:
https://github.com/prusa3d/PrusaSlicer/issues/1089#issuecomment-410262095

dahuafschmied, 2020-03-21:
I am talking about jerk settings m205 not m204 acceleration as in #1089

bubnikv, 2020-03-21:
@rtyr @DRracer ?

rtyr, 2020-03-21:
It is firmware issue, so it should be discussed here https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/292. Junction deviation is used only temporarily. Closing here.

closed 2020-03-21 by rtyr

dahuafschmied, 2020-03-21:
ok. opend the ticket here. no reaction on firmware ticket... :-(

rtyr, 2020-03-21:
I understand. We found a bug in classic jerk, it needs to be fixed by FW team.

dahuafschmied, 2020-03-21:
thank you for information!