prusa3d/PrusaSlicer #3879: Warn in case of too many retraction

espr14, 2020-03-21:
v2.2.0 rc5 win64

### Motivation
Too many retractions and deretractions at the same place on the filament lead to rubbing filament off by extruder wheels. Then the filament stops to move at all. FW is not able to detect it so the printed model can be thrown away.

Slicer can detect such situations because it can enumerate how many times each part of filament went through the wheels.

### Solution
Add an expert setting to filament category defining maximal number of passing through the wheels. A checkbox can be added to disable this intensive computation.

Number of passes can be done analytically or numerically. Since the length of retraction is short, typically, the numerical solution with 0.2 mm precision is quite easy and fast to implement.

bubnikv, 2020-03-21:
https://github.com/prusa3d/PrusaSlicer/issues/2216
https://github.com/prusa3d/PrusaSlicer/issues/3612

closed 2020-03-21 by bubnikv

bubnikv, 2020-03-21:
As you see similar issues were discussed.
I can assure you I am aware of these possibilities.
I am sorry, but just processing your high rate of new proposals (I can generate these proposals with such a high rate as well) puts quite a high strain on the developers just reading them, trying to understand what your intentions were, then to search for similar proposals to link them together to make sense of the high volume of traffic here on github issue tracker. This all draws our energy away from making PrusaSlicer better.