prusa3d/PrusaSlicer #3880: Incorrect purge volume when using a raft and soluble support

Lukrative525, 2020-03-22:
### Version
2.2.0

### Operating system type + version
Windows 7 Professional

### 3D printer brand / version + firmware version (if known)
Custom i3 running RepRapFirmware

### Behavior

_Describe the problem:_

I set up a print job to use soluble support material, a raft, and a wipe tower, where the interface layers of both the raft and the support are printed out of the soluble filament. The first filament change doesn't purge as much filament as the rest, which can lead to issues in the stability of the wipe tower as well as the quality of the interface layer of the raft, which can be a big problem if there is only one interface layer.

_Steps needed to reproduce the problem:_

Load a model and enable soluble support, raft, and wipe tower. Select the extruder with the soluble filament for the interface layers only.

_Expected Results:_

All layers of the wipe tower where filament changes are present purge the specified volume of filament.

_Actual Results:_

The first time a filament change occurs, the purged volume is less than that specified.

_Problem Layer:_

[image]

_Subsequent, Correct Layers:_

[image]

_Is this a new feature request?_

No

#### Project File (.3MF) where problem occurs

[Mouthpiece Wipe Tower Issue.zip](https://github.com/prusa3d/PrusaSlicer/files/4364813/Mouthpiece.Wipe.Tower.Issue.zip)

JohnnyDeer, 2020-03-24:
Hi, issue is reproduced and confirmed. You can avoid this behavior with using "no sparse layer (EXPERIMENTAL)" function.

Lukrative525, 2020-03-24:
I tried enabling "no sparse layers (EXPERIMENTAL)" and although it does reduce the layers of the wipe tower as expected, the layer of the wipe tower which includes the first filament switch still shows a reduced purge volume.

lukasmatena, 2020-03-24:
In fact, I believe there is no issue, at least not with the wipe tower. The raft is printed at layer height 0.3 mm, white the object at 0.15mm. So to purge the same volume of filament, shorter extrusion is enough in the former case. And I believe you only compared the area of the wiping region on the wipe tower, not the volume itself.

Lukrative525, 2020-03-24:
Can we make this a feature request to be able to change the raft layer height then?

Lukrative525, 2020-03-27:
I'm closing this issue since it's not really an issue, and submitting a feature request instead.

closed 2020-03-27 by bubnikv

bubnikv, 2020-04-01:
The raft is printed with a maximum allowed layer height, which can be set at the Extruder page. You have zeros there, which means to use a default value, which equals to 3/4 of the nozzle diameter. For this particular print, you may set the maximum layer height of an extruder to 0.2mm and you will be golden.

bubnikv, 2020-04-01:
So the above does not work, the raft layer height is not limited, sorry.

Lukrative525, 2020-04-01:
It would seem that the raft height is being directly determined as 0.75% of the nozzle diameter. When I change the nozzle diameter to 0.5mm, for example, the raft height changes to 0.375mm.