prusa3d/PrusaSlicer #3881: Object mirrored bug

hoegge, 2020-03-22:
### Version
2.1.1

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
Windows 10 1909

### 3D printer brand / version + firmware version (if known)
Prusa mk2.5.s
### Behavior
I printed an object an saw it came out mirrored. When I load the project, the object IS actually mirrored. I have never used the mirror function and it is two levels down in the right click menu so you don't do it by accident.

Is there a shortcut I could have touched by accident or is this a potential bug?

#### Project File (.3MF) where problem occurs
n/a - available on request

bubnikv, 2020-03-22:
Could you reproduce it? By the way, we released PrusaSlicer 2.2.0 final today, so you may want to try reproduce with it.

hoegge, 2020-03-22:
Hi
No real way to reproduce it. Have tried to load other .3mf files created on another PC to see if that was the issue, but no luck. 

IS there any mirror keyboard shortcuts or mouse gestures or anything, I could have hit by mistake?

rtyr, 2020-04-19:
Closing. Feel free to reopen it if it happens again with 2.2.0.

closed 2020-04-19 by rtyr