prusa3d/PrusaSlicer #3882: Feature Request: Printer Instances

Avatarsia, 2020-03-22:
Hello, 
is it possible to add tab's like in a browser so you can open multiple instances of prusa slicer without have 100 windows open ? 
also an option to safe these instances (automaticly when closing) and reload them when you reopen prusa slicer would be amazing.
so an windows update never again let your work vanish

PurchenZuPoden, 2023-01-28:
I was looking for a solution and found your feature request. I would love to have this feature in prusa slicer!

mcstoufer, 2023-07-27:
The 3MF spec claims it can have one root and multiple models. It would be useful to leverage that and have each tab be a component/model.
My use case is having to deal with going between multiple windows when I'm working on the same assembly with multiple parts.

Having multiple tabs that span across multiple logical builds is useful if there was some "grouping" ability here as well.

uberthin, 2023-08-18:
I would find this very useful.  I'm frequently working on multiple projects at once, but I don't want the clutter of many open windows.

Kammi61871, 2024-08-19:
is there any progress here? it's an usfull feature.

lukasmatena, 2026-09-01:
Implemented in [3.0.0-alpha11](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_3.0.0-alpha11). Closing.

closed 2026-09-01 by lukasmatena