prusa3d/PrusaSlicer #3885: Getting rid of Seam Blob

FlailAway, 2020-03-22:
### Version
2.2.0-rc2

### Operating system type + version
Mint 19.3+Cinnamon

### 3D printer brand / version + firmware version (if known)
Gen-L, TMC2208
Marlin 1.1.9

### Behavior
I have a 2" x 1" pipe-bush threaded inside and outside and I am getting random blobs on the thread tips. I assume these are Seam-blobs.

Retraction is 6.5mm at 40mm/s
Seam I have tried at Random and Nearest but nothing changes.

If the blob is totally unavoidable in PS, then it would be OK if the Seam blob was at the Root of the thread. Is there some way I can do that?

I hate bringing up Cura, but I managed to get the Seam-blob to not happen at all after watching Chuck's (CHEP on YT) video on it.

pilotltd, 2020-03-30:
Lower deretraction speed in printer settings/extruder1 from default of same as retraction speed.

rtyr, 2021-02-09:
>  it would be OK if the Seam blob was at the Root of the thread. Is there some way I can do that?

It is possible to paint seam locations since 2.3.0. 

If you have some specific suggestions for seam hiding improvements, you can discuss it here https://github.com/prusa3d/PrusaSlicer/issues/281.

Closing.

closed 2021-02-09 by rtyr