prusa3d/PrusaSlicer #3888: Configuration assistant owerwrites the user printer settings with the same name

DominikPalo, 2020-03-22:
### Version
2.2.0+

### Operating system type + version
macOS 10.15.3

### Behavior
I had my own printer settings called "Creality Ender-3". Then I added a new printer using Add/Remove printers functionality form the Printer dropdown - I selected Creality -> Ender-3. After clicking to the Finish, my own settings have been overwritten without any warning with the new settings also called "Creality ENDER-3". Now I have just "Creality ENDER-3" item under "System presets", but my old "Creality Ender-3" settings are missing and are not displayed under "User presets". I think there should be at least a warning before overwriting user settings (if it is not possible to have settings with the same names under System presents and also under User presets)

YuSanka, 2020-03-23:
Thanks for your report. I can reproduce this issue on Windows too

lukasmatena, 2024-02-19:
SPE-2166

closed 2026-09-29 by prusaslicer-bot[bot]