prusa3d/PrusaSlicer #3889: It is not possible to switch to the user printer preset with the same name as a system preset

DominikPalo, 2020-03-22:
### Version
2.2.0+

### Operating system type + version
macOS 10.15.3

### Behavior
Following the issue https://github.com/prusa3d/PrusaSlicer/issues/3888 I wanted to somehow restore my old settings, so I opened one of my 3mf files and saved printed settings from that file. Now I have two printer presets with almost the same name (just a different casing), but I'm not able to switch between these presenrs - when I click to the user preset, nothing happens - see attached screen recordings
 [image] and [image]

FidelCapo, 2020-03-23:
Thank you for report. I can reproduce it. Workaround is use different name for your user profile because I think same name as system profile name should not be accepted.

FidelCapo, 2020-08-05:
Fix for this issue will be part of next release.
Closing.

closed 2020-08-05 by FidelCapo