prusa3d/PrusaSlicer #3890: It is not possible to remove a system printer settings

DominikPalo, 2020-03-22:
### Version
2.2.0+

### Operating system type + version
macOS 10.15.3

### Behavior
* Click the "Add/Remove printers"
* Uncheck the printer you want to remove (in my case "Creality ENDER-3")
* The "Finish" button greys out (so it is not possible to save changes made in Configuration Assitant)
* Close the Configuration Assitant
* "Creality ENDER-3" item is still there

See [image]

rtyr, 2020-03-22:
It is probably macOS related, win64 version works fine for me.

bubnikv, 2020-03-22:
I suppose the issue is related to the name of the system profile equal to
the name of the user profile.

rtyr, 2021-02-09:
Fixed in 2.3.0. Closing.

closed 2021-02-09 by rtyr