prusa3d/PrusaSlicer #3891: Selecting Tool 2 does not heat correct tool with multiple extruders

turbo178, 2020-03-22:
Version

Version 2.1.1 + win64
Operating system type + version

Windows 10 Home V# 10.0.18363 Build 18363

In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog
3D printer brand / version + firmware version (if known)

DBot CoreXY with RepRap Firmware V# 2.05.1

### Behavior
I only want to use extruder #2 for a print, so i set all the values for extruder number to 2. when i send the print to my printer, it does not heat up the correct tool. i can see in the gcode that the heat command comes before the tool selection. i dont want to have to manually input the tool in the start code, in case i change my values in the settings. this is a multi extruder setup, (e3d chimera). 
* _Steps needed to reproduce the problem_
Select extruder 2 for all settings in multiple extruders tab, slice, then print. see screenshot

* _Expected Results_
the Correct tool heats up, in this case tool 2
* _Actual Results_
tool 1 heats up on standby and the print starts, trying to cold extrude out of tool 2
  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

[image]
[image]

Lukrative525, 2020-03-22:
This isn't the most elegant solution, but I get around this using a G10 command like this:

G10 S[temperature_[initial_tool]] P[initial_tool]

mbordoni, 2020-10-25:
I have similar problem with prusa mk3s + mmu2s.
On filament switch the new filament temp is set before the old filament is unloaded.
This cause a bad tip on old filament (specially in case of soluble pva filament use) and frequent mmu2s hanging.
Please postpone new temp set in gcode.

closed 2026-09-29 by prusaslicer-bot[bot]