prusa3d/PrusaSlicer #3893: PrusaSlicer Console --load my_config.ini doesnt overwrites the 3mf profiles.

tixenre, 2020-03-22:
PrusaSlicer 2.2.0

Windows 10  64 bit

I'm writing a script in Python 3.82 for batch file slicing to calculate cost and times. 

[](https://github.com/tixenre/Production-Manager/blob/devel/slicer%20copy.py)

My coding skills are basic and it is my first project. Anyways I'm able to slice a .stl from the command prompt on windows and from the script without issues. 

What I'm trying to do but  the --load .ini doest work
```
f = Path(r'3mf\Sonic.3mf')  ##file
p = Path(r'PrusaSlicer\presets\print\2.ini') ## print profile 0.2mm
i = Path(r'PrusaSlicer\presets\filament\PLA.ini') ## print filament PLA
pp = Path(r'PrusaSlicer\presets\printer\def.ini') ## printer mk3s
s = f'{slicer_console}  -g {f2} --load {p} --load {i} --load {pp}'
subprocess.call(s, cwd=cwd)
```

The .3mf is sliced, but the files profiles are not modified

How I ended up doing it
```
f1 = Path(r'3mf\Sonic.3mf') ## 3mf File
s1 = f'{slicer_console}  ---export-stl {f1}' 
subprocess.call(s1, cwd=cwd) ## export stl
f2 = Path(r'3mf\Sonic.stl') ## stl exported
p = Path(r'PrusaSlicer\presets\print\2.ini') ## print profile 0.2mm
pp = Path(r'PrusaSlicer\presets\printer\def.ini')  ## print filament PLA
i = Path(r'PrusaSlicer\presets\filament\PLA.ini') ## printer mk3s
s2 = f'{slicer_console}  -g {f2} --load {p} --load {i} --load {pp}'
subprocess.call(s2, cwd=cwd)
```

NOTE: My profiles.ini are in PrusaSlicer/presets

I'm not sure if I'm missing something in my code or it there is a real issue.
Any comment about would help me a lot and if I have to provide more information about the code let me know
Cheers,
Martin Santini

bubnikv, 2020-03-23:
>  The .3mf is sliced, but the files profiles are not modified

I think there is a misunderstanding, the files profiles are not to be
modified by PrusaSlicer when slicing.

PrusaSlicer loads all the configs in the sequence they are declared.
Then it loads the 3mf, then it applies the configs loaded before over the
configs stored in the 3MF.
At least this is how I understand the current source code. If it does not
work this way, then there is a bug.

tixenre, 2020-03-23:
> I think there is a misunderstanding, the files profiles are not to be
modified by PrusaSlicer when slicing.

Well, maybe the profiles that I pass are not used, I didn't mean that the PrusaSlicer profiles are modified. 

> PrusaSlicer loads all the configs in the sequence they are declared.
Then it loads the 3mf, then it applies the configs loaded before over the
configs stored in the 3MF.

I understand that if I load a profile in the command line, it should use it despite of the profile in the 3mf. I know that Slic3r was coded before the .3mf mainstream usage. If I would know C++ I would try to take a look and even propose a fix.
Maybe I'm wrong and there is another way to do what I'm trying to do.

bubnikv, 2020-03-24:
Fixed with 015a4a37dc8697725128d6e4107a5ea546b5c558 and c3a9915fa1fd02cd0954c2808f80280f679714f4

    The priority of config values is now:
    
    1) Config keys from command line, for example --fill-pattern=stars
       (highest priority, overwrites everything below)
    2) Config files loaded with --load
    3) Config from 3mf
       (lowest priority)

closed 2020-03-24 by bubnikv