prusa3d/PrusaSlicer #3896: Fontsize barely readable

BNoiZe, 2020-03-23:
### Version
2.2.0+

### Operating system type + version
macOS Catalina 10.15.4

### 3D printer brand / version + firmware version (if known)
Original Prusa MK3S / 3.9.0-RC1-3272

### Behavior
* On a 27 inch display with 2560x1440 resolution some fonts are so small that they are barely readable. The smallest textsize is found on the layer slider. Please see the attached screenshot.
[image]

#### Project File (.3MF) where problem occurs
Not related to a project file.

yetisnack1, 2020-12-15:
Yes, please fix this!  The slider is very difficult to see on anything other than a 1920 x 1080 monitor.

YuSanka, 2020-12-16:
Please retest with the latest version of PrusaSliser. This issue should be fixed with https://github.com/prusa3d/PrusaSlicer/commit/bca000a739b4cd328fdf9c92ac66ad2e9ece54eb

rtyr, 2021-02-09:
Is it ok with PS2.3.0? Can we close the issue?

yetisnack1, 2021-02-10:
> Is it ok with PS2.3.0? Can we close the issue?

Yes, thank you very much.

BNoiZe, 2021-02-10:
> Is it ok with PS2.3.0? Can we close the issue?

Yes. Fontsize ist way better!

closed 2021-02-10 by BNoiZe