prusa3d/PrusaSlicer #3898: Using rotate from command line results in object partially inside bed

albertoxamin, 2020-03-23:
For a University project in genetic algorithms I'm trying to find the best orientation of stl files.
However when I pass the rotate args, the object gets clipped in the bed. Is there any way to force the object on top of the bed from command line?
[image]

bubnikv, 2020-03-23:
I think you posted into a wrong repo. You likely wanted to post to the upstream slic3r

albertoxamin, 2020-03-23:
I posted it also on the slic3r repo, but I also posted this here hoping that someone knows a workaround

bubnikv, 2020-03-23:
did you test with prusaslicer?

albertoxamin, 2020-03-23:
Yes, I'm using PrusaSlicer
[image]

albertoxamin, 2020-03-24:
I managed to solve it by using `--align-xy 0,0`

closed 2020-03-24 by albertoxamin