prusa3d/PrusaSlicer #3899: Crash when trying to set up non-existent flashair

steken, 2020-03-23:
### Version
PrusaSlicer 2.2.0+

### Operating system type + version
MacOS High Sierra 10.13.6

### 3D printer brand / version + firmware version (if known)
Prusa MK3S

### Behavior
PrusaSlicer crashes, when I push the test-button, in the flashair setup. (printer turned off)

Printer profile: "Original Prusa i3 MK3S"
Printer settings - General - Host Type: FlashAir
Printer settings - General - Hostname, IP or URL:  ip or hostname to flashair-card, printer turned off.
Pressing the test-button makes PrusaSlicer crash.

If the printer is turned on, it works just fine.

FidelCapo, 2020-03-23:
Thank you for report. I can reproduce this.

bubnikv, 2020-03-24:
Fixed with ad5ae50933375f22337fc23dc3a930c8c672b8a5 on master.
We don't know when this fix will be released as part of an official build, but I suppose this is not a critical error, so the bugfix release may wait a bit.
Closing.

closed 2020-03-24 by bubnikv