prusa3d/PrusaSlicer #3900: Strange arrangement of 20 instances

DominikPalo, 2020-03-23:
### Version
2.2.0+

### Operating system type + version
macOS 10.15.3

### Behavior
* Add a small object to the plate
* Set number of instances of that object to 20
* Click "Arrange"
* 19 instances are arranged evenly, but 1 is misaligned

#### Project File (.3MF) where problem occurs
[matka_v3-arranged.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4371568/matka_v3-arranged.3mf.zip)

[image]

foreachthing, 2020-03-23:
Have a look at this similar issue: https://github.com/prusa3d/PrusaSlicer/issues/2824#issuecomment-599446793
Funny as h.e.-doublehockeystick.

tamasmeszaros, 2020-03-24:
Hi @foreachthing I see the problem, and I bumped into this recently but I had to postpone the fix due to more severe bugs for the latest release. I will get to this and investigate as soon as I can. Hopefully I can solve the irregular bed shape problem #2824  in one cycle.

timur-tabi, 2020-07-01:
I've attached another example.  This is supposed to be 150 copies of a small leaf.

[image]
[ECOS_Leaf.3mf.zip](https://github.com/prusa3d/PrusaSlicer/files/4860716/ECOS_Leaf.3mf.zip)

closed 2020-11-20 by tamasmeszaros

timur-tabi, 2020-12-05:
Looks like the problem still exists with 2.3 beta2.

[image]

bubnikv, 2020-12-06:
@timur-tabi 
PrusaSlicer tries to place the objects around the center, from that point of view this result is correct.

commit 096d0c98 (2020-11-20): Fix arrange objectfunction to avoid last object in awkward positions.

fixes #3900