prusa3d/PrusaSlicer #3901: PrusaSlicer 2.3.0 - Language Portuguese (Brazilian) dont work

ARCH-3D, 2020-03-23:
Hi,
After installing the new version PrusaSlicer 2.3.0, i try to put the language in Portuguese (Brazilian) on the Configuration Menu, but it put the language in English. The Portuguese (Brazilian) dont work, is a bug?

FidelCapo, 2020-03-23:
I cannot reproduce this issue. Can you please tell me more informations? What is your system and system language? Thank you

ARCH-3D, 2020-03-23:
Thanks FidelCapo. My system is windows 10, 64-bit, system language portuguese (Portugal).

bubnikv, 2020-03-24:
It may be a bug, but it may be difficult to reproduce. Could you please
make screenshots of your language settings in Windows 10, so we can
reproduce your setup?

ARCH-3D, 2020-03-24:
[image]
[image]
[image]
[image]

Thanks bubnikv. Do you need more?

ARCH-3D, 2020-03-24:
[image]

This also does not look normal 🤔

bubnikv, 2020-03-24:
that version number looks normal, it is shorter on purpose.

ARCH-3D, 2020-03-24:
So But shouldn't it be 2.3.0 instead of 2.2.0?

ARCH-3D, 2020-03-24:
Note: I tried to uninstall and install again, but the problem continues. It is always in English. In other languages it works

FidelCapo, 2020-03-24:
I can reproduce this issue when system language and format is set to Portuguese like on screenshots.

[image]
[image]

ARCH-3D, 2020-03-24:
It will be fixed in the next version?

bubnikv, 2020-03-26:
Fixed with a1aee69c5a501706a5ff3f98e2ed87742224a08b, but  it will likely take some time before the fix is released.

For now the simplest for you is likely to rename the directory resources\localization\pt_br to pt.

Closing for now.

closed 2020-03-26 by bubnikv

ARCH-3D, 2020-03-26:
Yes, it works that way :) Thank you bubnikv