prusa3d/PrusaSlicer #3902: Feature request > Save MMU selected filaments in printer profile

NielschB, 2020-03-23:
### Version
Prusaslicer 2.2.0_rc5

### Operating system type + version
Windows 10 x64

### 3D printer brand / version + firmware version (if known)
Prusa MK3S MMUxS ;)

### Behavior
I would like the selected filament for each extruder to be saved in the printer profile. As I have 18 filaments loaded on my MMU, each time I switch printer profiles the filament selection for all extruders is reset to one default filament, so I have to specify them over again.

When I export a config bundle I do see at the end of the file the assigned filaments, but this is just a workaround as I need to update the exported configs with every change I make to whatever settings.

ErikV1000, 2020-12-26:
I'm using my MK3S/MMU2S with a pretty defined set of fillaments loaded, however every time a 3mf file is sliced, the printer settings need to be reset to the actual situation on the printer (which did not change). In addition when opening an existing MMU 3mf file a new printer profile is loaded, but the filaments are no longer linked to the objects on the bed, these behaviours may be seperarate in the software, from a user experience its part of the same "activtity".

When opening a stored MMU-3mf file the printer profile changes to the one stored in the 3mf file, with its own fillament definitions. Having made names and definitions for the filaments at first use, it is confusing to see "3mf" definitions mixed with "original" definitions. Especially because they are no longer linked to the objects on the bed. At the same time in the printer settings, the "reset to fillament color" feature has been disabled again. All in all when reworking an existing MMU-3mf file, the whole printer settings need to be redefined.

The same is for making a new MMU-3mf file on the existing printer configuration.
In addition, when switching between my Prusa MK3S-MMU2S and MK3S, all information is lost and the set-up needs to be redone completely (interestingly some of the generated fillament profiles remain available).

1) It would be great if fillaments could be stored in a central location, linked to the printer configuration, in stead of having multiple copies of the definition in different 3mf files.
2) Would it be possible to set the extruder color to fillament color as a default, in stead of whatever PruseSlicer comes up with? In case #2626 the option to go from extruder color to undefined was discussed and of course there is a "reset to fillament color" button, still somehow this information seems to get lost between runs.

[image]
Fillament set-up from 3mf file, taking the printer profile form the 3mf
[image]
Same Fillament set-up, but now switched back to the Prusa printer profile (thus releasing the extruder/color link)
[image]
Extruder color menu item I'm referring to.

All in all it would be great if some of the recurring correction tasks could be automated.

NielschB, 2021-02-15:
With slic3r  version 2.3.0 still the filament selections are reset when switching between printers.
Is there any indication that this will be fixed?

NielschB, 2022-04-10:
I notice that the filament selection for the extruders is saved in the generic PrusaSlicer.ini file, if it would be saved for each printer seperately in the .ini, this issue would be solved, anybody who can fix this?

rgeorgi, 2026-08-29:
This was driving me crazy switching between my MMU3 MK4S and a Prusa Mini, decided to do something about it. If you're able to compile your own version, the above PR addresses this problem and works for me in a modified 2.9.6

Wegerich, 2026-08-29:
In the end my workaround for this was to switch to OrcaSlicer

closed 2026-09-29 by prusaslicer-bot[bot]