prusa3d/PrusaSlicer #3905: No longe able to select the extruder in 2.2.0+

UltiCa, 2020-03-23:
Version:   2.2.0+ and 2.2.0 rc5
Build:     PrusaSlicer-2.2.0+-202003211132
Operating System:    Macintosh
System Architecture: 64 bit
System Version:      macOS Catalina Version 10.15.3 (Build 19D76)

[image]

### Behavior
A multimaterial object is loaded (as you can see)
I can not change any longer the extruder. The drop down field is no longer displayed.

_Is this a new feature request?_
No, in the last version, this was working

YuSanka, 2020-03-24:
Thanks for your bug report. For some reason we can't reproduce your issue. Which version do you mean by the 'latest'? Is it 2.1.0 or 2.2.0-rc5? If you are talking about 2.1.0, then the control was changed to use colored rectangles for each extruder. And because of an wxWidgets issue on OS X we can't use the native control. Please, use right mouse click to edit item's cells.
[image]

UltiCa, 2020-03-24:
Sorry, right mouse click is working to activate the extruder selection.
(I have tested Version: 2.2.0+ and 2.2.0 rc5
Build: PrusaSlicer-2.2.0+-202003211132)

closed 2020-03-24 by UltiCa