prusa3d/PrusaSlicer #3906: Values not being kept after changing

eboston, 2020-03-24:
### Version
_Version of PrusaSlicer used goes here_
PrusaSlicer-2.1.1+win64

_Use `About->About PrusaSlicer` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_
Doesn't matter.  Not an issue with the printer.

### Behavior
* _Describe the problem_
On Printer Settings page.  Multiple extruders.  Unable to save changes to retraction length when 
tool is disabled.

* _Steps needed to reproduce the problem_
Enter a value of zero (0) and move to a new field

  * _If this is a command-line slicing issue, include the options used_
n/a

* _Expected Results_
Expect value to remain as what was entered.

* _Actual Results_
Value reverts back to original value.

  * _Screenshots from __*PrusaSlicer*__ preview are preferred_

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
_Upload a PrusaSlicer Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for PrusaSlicer, Slic3r PE 1.42.0-alpha and newer)_
_Images (PNG, GIF, JPEG), PDFs or text files could be drag & dropped to the issue directly, while all other files need to be zipped first (.zip, .gz)_

Tried uploading the .3MF file, but it says it doesn't support the file type.

JohnnyDeer, 2020-03-24:
Hi, unfortunately i'm not able to reproduce it. Can you please try it on latest version 2.2.0?

bubnikv, 2020-03-24:
Also please test with https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.2.0

FidelCapo, 2020-03-24:
I reproduce it on win10 in version 2.1.1 with MK3 MMU2 and Extruder 4 where retraction length cannot be set to 0. I thing this is fixed in version 2.2.0. Can you please try it in this 2.2.0 version? Thank you.

rtyr, 2020-04-15:
No response. We believe that this is fixed in 2.2.0. I also tried it and I was unable to reproduce it. Closing.

closed 2020-04-15 by rtyr