prusa3d/PrusaSlicer #3911: Print paused message sticks on LCD when stopping print during pause

Windscreen2011, 2020-03-25:
### Version
2.2.0

### Operating system type + version
Windows 7 SP1 64 bit

### 3D printer brand / version + firmware version (if known)
MK3s  3.9.0-RC1-3272

### Behavior
If a print is stopped during M601 (pause print via layer selector), "print paused" sticks on LCD screen.

Windscreen2011, 2020-03-25:
Also, LCD menu would indicate that the print is truly cancelled in this state, but when powering the printer down, it goes into power panic mode.

JohnnyDeer, 2020-03-25:
Hi, this seems to be firmware problem, please close issue here on slicer and report it here https://github.com/prusa3d/Prusa-Firmware/issues , thank  you.

Windscreen2011, 2020-03-26:
Bug report written on Firmware site
https://github.com/prusa3d/Prusa-Firmware/issues/2558

closed 2020-03-26 by Windscreen2011